Klipper3d/klipper #2258: Bed mesh z correction not as expected (offset of about 2mm during print necessary)

AlmightyFrog, 2019-12-08:
Not sure what the issue is, guess it is on my side but I just can't find the problem.

I am using a FSR-Probe which i put directly between nozzle and bed, so offset between nozzle hit bed and nozzle hit probe is quite small.
Probe has measured about 0.3mm thickness measured with digital caliper.
Doing "Calibrating probe Z offset" as per https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md does result in about 0.15mm thickness which seems about to be fine as that FSR needs to be squished to trigger.
Printer is mechanical quite stable and probe accuracy test seems fine. So I guess probe is fine, this is what is added to printer.cfg after save:
```
#*# [probe]
#*# z_offset = 0.156
```

Next up, I wish to calibrate bed as this is a fixed one (without springs) and can't be calibrated at all. It is also not heated so never mind thermal expansion for now. As there is a mirror glass put on top i expect it to be quite flat.
I did call `BED_MESH_CALIBRATE` followed by `SAVE_CONFIG` after all 9 pints were successfully probed using my FSR-Probe.
Now following is added to my config which is totally off:
```
#*# [bed_mesh default]
#*# points =
#*# 	  1.278815, 2.084037, 2.804499
#*# 	  1.216875, 2.108487, 2.871329
#*# 	  1.146785, 2.077517, 2.881109
```
Expected range would be as per my understanding between -1 and +1 as this is relative to z_offset, right?

I can see Z axis working when i now move e.g. X axis, so i guess bed mesh is working and used.

Odd is following:
I turn on the printer and home. Now it shows on Display in Move-Window "Z:-01.8" and nozzle is roughly 0.3mm away from touching bad. So far so good.
Now when I alter Z to be 0, Display in Move-Window shows "Z:000.0" as expected, but nozzle does not go further down on bed but rather up in the air to about 1.9mm (hard to measure, used wooden stick which about fits perfectly in between to get this number).

If I now manually do set `SET_GCODE_OFFSET Z=-2` during printing print works as expected. I can also see the Z axis working here when X and/or Y axis are moving during print moves.

Here is the klippy.log: [20191208-bedlevel-fail-klippy.log](https://github.com/KevinOConnor/klipper/files/3937259/20191208-bedlevel-fail-klippy.log)

Also I've create a M112 dump, just i case this might help: [20191208-bedlevel-fail-M112.log](https://github.com/KevinOConnor/klipper/files/3937265/20191208-bedlevel-fail-M112.log)

Note: Yes, my bed is currently about 1.5mm tilted which is not good for precise printing. This is not part of the issue and will be eventually solved at some later time when I disassemble the synchronization axis between the two Z-Axis belts and try to align it better..

Arksine, 2019-12-09:
If we assume that you are homing toward the front left corner, that corner's probed value is Z=1.27.  Your position endstop on z is -.5, so after homing the printer thinks that corner is at gcode position z=-.5.  When you move to Z=0, you would naturally move .5 up to reach zero, then bed_mesh adds the 1.27mm of adjustment leading to ~1.8mm as you are reporting. 

In short, I believe that `position_endstop` in the `[stepper_z]` section is off.  I suggest doing Z_ENDSTOP_CALIBRATE, then re-probe the mesh.

AlmightyFrog, 2019-12-10:
Seems like that was key to success. It is still a tiny little off, but that is only 0.2mm so that i can tweak by setting Z_ENDSTOP_CALIBRATE just a little more away from bed.

But anyhow, just that I got it right, could you please confirm if my understanding of correlation between parameters is correct:

Z Probe offset only needs to be calibrated only once and then will most likely never change (until i would change the probe of cause).
Q0: Z Probe itself has no influence on Bed Mesh?

Bed Mesh relies on one calibrated point to be "equal" with Z=0, this is first point in calibration, so in my case front left corner:
Q1: Does that mean, I should make the Z_ENDSTOP_CALIBRATE not in center of bed but at that front left corner?

Q2: So if I want to recalibrte my bed I can't trust of that Z_ENDSTOP_CALIBRATE is still fitting, so i have to always do both Z_ENDSTOP_CALIBRATE and BED_MESH?

KevinOConnor, 2019-12-23:
I can try to answer your questions:

Q0: No - having a correct probe z_offset is important for bed mesh.  As near as I can tell, your reports were based on some of the internal Klipper locations - which indeed will show negative/positive values depending on the coordinate system those locations are reported in.  As near as I can tell, with a bed mesh properly implemented, a move to "X123 Y456 Z0" would have actually moved the nozzle to a position ~0mm from the bed.

Q1: I would recommend running the endstop calibration at the center of the bed.

Q2: I'd recommend running both again.

-Kevin

AlmightyFrog, 2020-01-01:
Thanks for clarification.

closed 2020-01-01 by AlmightyFrog