Klipper3d/klipper #2259: Arduino Due + I2C OLED 128x64 - ssd1306

1walkerit, 2019-12-09:
hi
I am trying to connect i2c OLED dispaly and Arduino Due. Reading the last issue it could be working but I still have problem. I connect i2c to pins 20,21 (SDA,SCL)  and no success, Can you help me to solve the problem?
[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/3940537/klippy.log)

KevinOConnor, 2019-12-09:
Try:
```
[display]
lcd_type = ssd1306
i2c_mcu = mcu
i2c_bus = twi1
```

-Kevin

1walkerit, 2019-12-09:
> [display]
lcd_type = ssd1306
i2c_mcu = mcu
i2c_bus = twi1

GREAT!
it is working, thank you very much and, please, close this issue. (y)

closed 2019-12-10 by 1walkerit