Klipper3d/klipper #2260: Failed to home : Timeout during endstop homing after update

Jerommeke, 2019-12-09:
Hi Kevin,

first off all thank you for all your great work on Klipper.  I have been using it now for a year and it worked great on my cr-10 with mks gen-l board with tmc 2130's.

this weekend I upgraded the octopi and klipper host & firmware and since then I get the homing error, not being able to home means no printing.

I did read in the braking changes about the virtual endstops (which I use) but can't figure out what to change in the config.

according to the docs the config looks valid to me.

communication to the stepper drivers is working.

hope you will have a look at my attached klippylog.

Kind regard

Jeroen

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3941666/klippy.log.zip)

ld3300, 2019-12-11:
If you put the homing commands manually into the terminal does it time out?

I wonder if the timeout is coming from octopi, i'm not completely sure where the log files are generated. But since you are doing the g28 calibration in your gcode macros then the octoprint won't be aware to wait for it.  I'm guessing START_PRINT is a command coming from your slicer.  Try adding it to the "Long Running Commands" section of "Serial Connection" > "Firmware and Protocol" > "Advanced".  I don't know if octopi allows extended g-code, but there are a couple of other tricks we can try if that doesn't work.  For example we can return any unused gcode as a command and use action commands to drive it through the host and include that code in the long running commands.

Let me know if any of this helps out.

Jerommeke, 2019-12-11:
Hi id3300,

a G28 directly in the terminal has the same effect:

[...]
Send: G28
Recv: !! Failed to home x: Timeout during endstop homing
Recv: !! Failed to home x: Timeout during endstop homing
Recv: ok
[...]

I get the error after +/- 10 sec giving the command.

The START_PRINT does indeed come from the slicer, but I don't think this has anything to do with the problem since I get the error too by giving a G28 in the terminal.

thx for looking into in it.

jakep82, 2019-12-11:
Does the motor move?  If not, verify that your settings and wiring are correct.  STEPPER_BUZZ can help with this as described at the link.

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md#verify-stepper-motors

Jerommeke, 2019-12-12:
Hi jake82,

Thank you for pointing me in the right direction, all the motors worked except the x-axis because of a broken connection.  This had nothing to do with me updating it, coincidently it happend at the same time.  Happily printing again.

closed 2019-12-12 by Jerommeke