Klipper3d/klipper #2261: Printer times out

Avalonnw, 2019-12-09:
Hi,
My printer sometimes loses comms with host, with messages reporting clock timeout. It happens randomly and the same file can be reprinted after the issue. I have attached log.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3942283/klippy.zip)

ld3300, 2019-12-11:
I'm not seeing anything weird.  Does it always happen at the same point in the print? Have you tried a different USB cable?

Avalonnw, 2019-12-11:
@ld3300 Thanks for looking. It does not happen at the same point, and sometimes I would just restart the printer and start the same file and it would print ok.
I admit I haven't tried different cable as the one I'm using has been in use for a while.
It's a coincidence that the first time I got this sort of issue was after I moved the house AND updated klipper on both host and printer board.

KevinOConnor, 2019-12-13:
The best advice we have on this is at https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors .

Also, it looks like you've done a custom install of the software.  If so, note that the ModemManager OS process is known to cause these types of connection issues.

-Kevin

Avalonnw, 2019-12-13:
@KevinOConnor Thanks, I saw that FAQ section.
The thing is, untill I moved the printer and updated the software (including Raspbian) I've never seen anything like that. I was a bit distracted the last couple of days and did not check, but will swap the cable and see what happens.
When you mention "custom" install, what do you mean? You don't have pre-made images as far as I am aware, so I use a default Raspbian on a laptop and Octoprint.

KevinOConnor, 2019-12-16:
> When you mention "custom" install, what do you mean?

I mean that you are not running it on a Raspberry Pi.  That's fine, just be aware that ModemManager is often a culprit.

-Kevin

Avalonnw, 2019-12-16:
Ok, thanks. I was not aware there were restrictions like that.
I have re-connected all the cables, straightened up the usb cable and it seems to be working for now - 7h print went ok. Will keep monitoring.

closed 2019-12-16 by Avalonnw