Klipper3d/klipper #2264: cannot connect to printer -- unable to build c_helper.so

ndaman, 2019-12-11:
I'm not sure if these are the same or different issues, but I have installed klipper per the instructions and when I try to connect to '/tmp/printer' in octoprint I get the following:

```Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0xffff803cb290, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Send: N0 M110 N0*125
No answer from the printer within the connection timeout
```

So I checked /tmp/klippy.log, which says it could not build the chelper file:

```Starting Klippy...
Args: ['/home/nick/klipper/klippy/klippy.py', '/home/nick/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-187-g4cbbe18a'
CPU: 4 core ?
Python: '2.7.15+ (default, Oct  7 2019, 17:39:04) \n[GCC 7.4.0]'
Building C code module c_helper.so
Unable to build C code module (error=256)
```

Is the c_helper issue causing this printer connection problem? If so, how can I fix it? If not, what else can I check to see why I'm not able to connect to my printer? (I am using an Orange Pi Zero+ connected to an Ender 3 Pro, by the way)

BlackStump, 2019-12-11:
are you using armbian buster or bionic?
I am using a orangepi zero + with armbian buster with no problems

ndaman, 2019-12-11:
I was using Bionic. I was thinking I should start everything fresh anyway, so I'll give Buster a try tonight. Were there any particular directions you followed for octoprint + klipper?

ndaman, 2019-12-11:
I've attached klippy.log as an attachment to make the bot happy.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3951201/klippy.log)

ndaman, 2019-12-12:
Well, I'm not sure what the problem was, I did a fresh install (used Buster per @BlackStump  recommendation), but my buck converters also arrived today, so I am also powering the device differently than I used to be. Regardless, everything seems to be fine now, thanks!

closed 2019-12-12 by ndaman