Klipper3d/klipper #2266: make flash fails with ioctl("TIOCMGET"): invalid argument

manfre, 2019-12-12:
Updated to latest master (4cbbe18afdb44b08636d217eb9a63b466502da75) and no longer able to flash devices.

Spec Voron with RAMPS and Arduino Mega 2560.
Klipper on rpi 3b+

```
# ls /dev/serial/by-id
usb-1a86_USB2.0-Serial-if00-port0  usb-Arduino__www.arduino.cc__0042_9573033363735111A061-if00
# make flash FLASH_DEVICE=usb-Arduino__www.arduino.cc__0042_9573033363735111A061-if00
  Flashing out/klipper.elf.hex to usb-Arduino__www.arduino.cc__0042_9573033363735111A061-if00 via avrdude
avrdude: ser_open(): can't open device "usb-Arduino__www.arduino.cc__0042_9573033363735111A061-if00": No such file or directory
ioctl("TIOCMGET"): Invalid argument
ioctl("TIOCMGET"): Invalid argument
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
vrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.

src/avr/Makefile:32: recipe for target 'flash' failed
make: *** [flash] Error 1

```

manfre, 2019-12-12:
Empty log to silence bot and remove tag. Log is not relevant to this issue.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3953640/klippy.log)

make menuconfig attached. I've tried default Baud of 250000 and 115200.
[config.txt](https://github.com/KevinOConnor/klipper/files/3953647/config.txt)

FHeilmann, 2019-12-12:
your flash command is missing `/dev/serial/by-id/`. You need to provide the full path to the device, not just its ID

manfre, 2019-12-12:
*facepalm* Thank you, and sorry.

closed 2019-12-12 by manfre