Klipper3d/klipper #2268: MCU 'mcu' shutdown: ADC out of range

robustini, 2019-12-12:
I'm going crazy with this error and can't get away with it.
The board is a RuRamps 1.3, the pins should all be correct but Klipper continues to go to shutdown, from what I read for a problem I believe in one of the thermistors.
At the hardware level they work without problems with Marlin.
Thanks in advance to anyone who can help me.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3956101/klippy.log)

robustini, 2019-12-12:
I think I solved something, I had a wrong range on the bed.
Now the result is this:

[klippy.log](https://github.com/KevinOConnor/klipper/files/3956209/klippy.log)

mental405, 2019-12-12:
ADC Out of Range errors typically occur when the analog pin is floating because of a loose or disconnected thermistor connection. Another cause could be due to an improperly configured heater in the config. It looks like you resolved that part while I was responding with this.

In terms of your new error. Try issuing "FIRMWARE_RESTART" from the terminal and see what happens.

robustini, 2019-12-12:
FIRMWARE_RESTART from terminal does not produce any result, under Repetier Server (RPI3) apparently fails to reset Arduino Due.
I have to rewrite the firmware every time to force a reset.

robustini, 2019-12-12:
After start the service now i've this one in the log:

Starting heater checks for heater_bed
Enabling TMC virtual enable for 'stepper_a'
Enabling TMC virtual enable for 'stepper_b'
Enabling TMC virtual enable for 'stepper_c'
Starting heater checks for extruder

But from console:

15:56:27.654: // Klipper state: Ready
15:56:41.238: Firmware stopped! You can only send host and shell commands until you hit emergency stop or restart the printer.

robustini, 2019-12-12:
Ok, now it works, I have the temperatures!

closed 2019-12-12 by robustini

Bob90, 2020-12-15:
I got this error after a 30 seconds during a nozzle change at max temp (275). I lowered the temp to 250 and it worked fine. Is it to do with expected thermal profile for the heater? I had trouble with a PID for the extruder if my fans were too high too.

Steffeng5, 2021-04-22:
Had the same problem. Found this in the logfile:
MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

So: If max_temp is exceeded. (e.g. while heating going to 251 when 250 is max_temp, then it crashes.
Set max_temp to 270 and print with 250 works fine

Buktron, 2021-09-03:
Had the same problem. Found this in the logfile:
MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

I'm having the same problem with my Ender 3 V2 with the Phaetus Dragon Hotend ST and no matter how many times I change the max_temp is doesn't do anything at all. I'm not sure what to do. I've some lines for the necessary firmware to work according to popular YouTube videos but I just can't seem to get my printer working with this. I also have an orbiter extruder and even with it plugged in or not I still get the same issue. Any help would be appreciated.