Klipper3d/klipper #2271: [Feature Request] Plase add "Carriage" state (from SET_DUAL_CARRIAGE) to SAVE_STATE

w1ebr, 2019-12-13:
Hello Kevin, Would you please add the value of "carriage" to the saved (and restored) states?

KevinOConnor, 2019-12-13:
Alas, storing the current carriage in the "gcode state" is not something that is straight forward to do.  It's also not part of the "gcode parsing state" and thus I'm not sure it is a good place to store it.  It is possible to store state in macros via the SET_GCODE_VARIABLE command (and similar) though.

-Kevin

w1ebr, 2019-12-13:
Thanks for your quick response above!
Is there a way to access a variable defined in one macro ( like one whose value is changed using SET_GCODE_VARIABLE ) so that I can use its' value in a different macro? The problem I am trying to solve is how to save carriage state so that I can move the carriages between layers to do a "nozzle wipe". If I created a macro which defined a variable called "current_carriage", I could set its' value in the "T0" and "T1" macros that change which carriage is in use. I would like to be able to use this value in "wipe_nozzle", "park_t0", "park_t1", "enter_dumpzone" and "clear_dumpzone" macros where I would save the current gcode state, save the current carriage in use, move the carriages as required and then restore the gcode state and the carriage that was in use. The alternative is to create more complex slicing scripts.

KevinOConnor, 2019-12-16:
See the docs at: https://www.klipper3d.org/Command_Templates.html

-Kevin

w1ebr, 2019-12-16:
Great! Thanks!

closed 2019-12-16 by w1ebr