Klipper3d/klipper #2272: Must home axis first: 0.000 0.000 19.990 [0.000]

dpackham, 2019-12-13:
I get this on the screen when I try to issue a G8 or G29 via CLI
I am running the latest code via git pull as of this morning.  
I reflashed Marlin FW to test this and the printer does work and home. 
Any ideas?   Thanks

[printer.zip](https://github.com/KevinOConnor/klipper/files/3961530/printer.zip)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3961524/klippy.zip)

08:43:53 Must home axis first: 0.000 0.000 19.990 [0.000]
08:43:53 Must home axis first: 0.000 0.000 19.990 [0.000]
08:43:54 Must home axis first: 0.000 0.000 5.095 [0.000]
08:43:54 Must home axis first: 0.000 0.000 0.100 [0.000]
08:43:56 Must home axis first: 0.000 0.000 5.095 [0.000]
08:43:56 Must home axis first: 0.000 0.000 19.990 [0.000]
08:43:56 Must home axis first: 0.000 0.000 19.990 [0.000]

tonn0297, 2019-12-13:
g28 first to home. 
klipper dont have a g29 you will have to make that macro. look at exsample_macro

dpackham, 2019-12-13:
G28 doesn't even work.  Do I have to define that as well?

Hywelmartin, 2019-12-14:
Your homeing override might be faulty.. I have never used it....

dpackham, 2019-12-14:
Yup that was it...   something must have changed because it was working.   the one you referenced does work and am now using it.

closed 2019-12-14 by dpackham