Klipper3d/klipper #2274: Feature request: TMC5160 StealthChop Automatic Tuning

kookaburra88, 2019-12-14:
After switching the x&y Axis motor from 1.8° to 0.9° units, I found StealthChop to produce more resonance/to be louder than SpreadCycle mode. With the 1.8° steppers, StealthChop was running super quiet as expected.

According to the TMC5160 datasheet (file:///C:/Users/Christopher/Downloads/TMC5160A_Datasheet_Rev1.13.pdf, p. 57) the drivers provide automatic tuning procedure, to adapt to the motor paramaters. Has this been tested and implemented here?

KevinOConnor, 2019-12-16:
A developer would need to implement and submit it.

FWIW, I'm not aware of any way to auto tune the trinamic drivers - as far as I know tuning requires external probe hardware to perform measurements.

-Kevin

kookaburra88, 2019-12-17:
Ok, I will need my time to work though the dataset, If someone has experience with configuring the TMC drivers, please let me know.

regards

Spacemarine2018, 2019-12-28:
@kookaburra88 

Did you make any progress? Can you already tell, what could be tuned automatically if klipper would support it?

kookaburra88, 2019-12-28:
@Spacemarine2018 Well...if I understood the datasheet correctly, Stealthchop optimization is practically triggered after homing the printer and then keeps optimizing on and on.

Interestengly I found that Stealthchop optimization on TMC2208 seem to work fine (1.8° & 0.9° steppers), since 
- Stealthchop is less noisier than spreadcycle (for TMC5160 it's vice versa)
- TMC parameter change during printing (this is also the case for TMC5160).

If there is someone with TMC5160, please check whether Stealthchop is silent compared to Stealthchop with klipper and report your config?

KevinOConnor, 2020-01-03:
Yes, as I understand it, stealthchop is largely "self tuning" and the 3d printer software does not need to do anything particularly special.  I don't have any insight as to why tmc2208 would differ from tmc5160.

-Kevin

KevinOConnor, 2020-01-23:
I'm going to close this as I think the original question was answered (to the best of our knowledge, no special tuning is needed for stealthchop).

-Kevin

closed 2020-01-23 by KevinOConnor