Klipper3d/klipper #2276: Print stops, Pi hangs

mediocre-to, 2019-12-14:
Hi.
After a few hours of printing the Pi hangs, Octo goes offline, SSH attempts fail and the printer stops and the nozzle gets stuck in the print. The printer i/f doesn't respond to the knob. Happened several times now. First I guessed on a temperature issue and arranged for more cooling of the Pi. Afterwards it reports around 40 deg(C) on idle and 45 on printing. Memory usage goes from 16/85 at octo restart to 31/85 but stays there. Processor is utilized between 5 - 12 % during print. 

But it still crashes. I manages to retrieve the klippy.log from the EXT fs on my PC, without a Pi-restart. It shows several attempts from my side with lower baud-rates than default 250 000 between klipper and the printer and after giving that up I started a print that crashed. 
[klippy2.log](https://github.com/KevinOConnor/klipper/files/3964257/klippy2.log)

I am also attaching an older log-file that was retrieved after a power off/on-restart after the crash before. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3964259/klippy.log)

I am not sure this is a klipper issue, but ... Help ... anyone?

BR Torsten

tonn0297, 2019-12-14:
that sounds to me as undersized psu for the pi. if pi3 then minium is 2.5A. if pi4 minimum is 3.5A-

mediocre-to, 2019-12-14:
Pi-PSU has the Pi-logo, the cable is fixed (No USB-out) and it states 5V 2.5A. Octo does not report undervoltage. Pi is 3B+ 

Still undersized?

Hywelmartin, 2019-12-15:
bad sd-card on the pi??

mediocre-to, 2019-12-15:
Eventually ... Sandisk Ultra 32GB, newly "etched" octopi partitions without issues.

mediocre-to, 2019-12-17:
Ok. So I continued the search for reasons for the unwanted behaviour.

1. Rpi 2 with USB-dongle wifi and another 16GB SD-Card. OK!
2. Rpi 3B+ with the above mentioned card. OK!
3. Fsck recover on the 32GB card from the failed print. Some errors fixed, mainly lost inodes. Ran a non-destructive read/write test, no bad blocks discovered.
4. Rpi 3B+ with the recovered 32CB Card. Fail! Octoprint webinterface "server seem to be offline". The printer dead on the display, not responding to the wheel/knob and the extfan on full speed although the extruder was cold. (As with previous failures)

But this time the SSH session was still alive. ~30% processor utilization and temp 52 deg C. Ran ps aux and top. Outputs enlosed. 
[psa.txt](https://github.com/KevinOConnor/klipper/files/3975491/psa.txt)
[top.txt](https://github.com/KevinOConnor/klipper/files/3975492/top.txt)

Restarted the Octoprint service, Processor util normal again and webinterface active. Connect to klipper and a FIRMWARE_RESTART restores the printer to normal. 

I am also enclosing the klippy.log that had grown rather big.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3975503/klippy.log)

Much of what has happened points to the SD-card being unreliable (Hywelmartin suggested) but ... from the process list it seems octoprint is doing a lot, but not anything useful and since the klippy.log seemingly was updated after the print was stopped. 

I am not a programmer, only a happy amateur googling around for solutions. 

What buggers me is that octoprint seem to do something useless forever. Shouldn't octo be able to recover?  

I would like someones opinion before posting an octoprint issue, while I have this thread open.

BR Torsten

CraXyOW3, 2019-12-22:
Have a very similar problem.
Starts with a "MCU 'mcu' shutdown: Timer too close" and then freezes, no response on knob or anything.
Once i sat next to the printer when it happened and i got the klippy log, ssh and octoprint working.
Then it froze and i had to restart Raspberry.

Tried with same SD (Toshiba Excercia 16GB UHS1 aka class10 ) card 2 times on 2 separate RPI's, one RPI 3B and one RPI 3B+.
I also set the virtualSD card, no change.
Will try to replace SD Card with a fresh Octoprint/Klipper install.

Usually i can print one session, then it happens.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3992781/klippy.log)

CraXyOW3, 2019-12-24:
With the new SD card i had no more errors. So i am to no further use.
This is with new octoprint install for a RPI 3B+, addon thespaghettidetective and octoscreen.

jakep82, 2019-12-24:
> What buggers me is that octoprint seem to do something useless forever. Shouldn't octo be able to recover?
> 
> I would like someones opinion before posting an octoprint issue, while I have this thread open.
> 
> BR Torsten

If the card is failing, Octoprint might be trying to access a corrupt file which would cause major performance issues.  If a new card fixes the issue, then I don't think you have anything to report.  This appears to be strictly a hardware issue.

mediocre-to, 2019-12-30:
Ok, no more comments. I have tried to decipher the klippy.log and I am still unsure if it is a SD-card problem or an unreliable mcu. I have not been using the printer, I also have e rather crappy cartesian build with a duet board. It does what it iss supposed to do, when the hardware hold together.

I will get back to klipper when my patience and curiosity comes back.

/T

closed 2019-12-30 by mediocre-to