Klipper3d/klipper #2279: BTT SKR v1.3 with TMC2208v3 --  Unable to flash

jasonshrout, 2019-12-15:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3965170/klippy.log)

Kevin and Fellows/Ladies,

Long time advocate, first time posting.  Recently (2 weeks ago), I made the decision to buy and install the BTT SKR v1.3 with 5 TMC2208 version 3's.  Having had so much success with Klipper on the ANET A8, it is the only firmware that I feel comfortable using and I sincerely want to use it on this 32bit board.  But...  despite following the instructions and trying other suggested voodoo methods, this project is defying me.  I absolutely do not want to use Smoothie, nor Marlin.  Unfortunately, those two actually allow communication with the hardware and I can move my steppers, read temps, but homing is a lost cause... which is another reason for my resolve to only use Klipper.

I can only assume that the bootloader on this board is one of the newer ones that you mentioned in August that everyone is having so many problems with.  

The following have been completed:
upgraded to most recent Octopi
pulled and reinstalled the latest Klipper
verified multiple good SD cards
verified that all klipper.bin have been renamed to firmware.bin when moved to SD Card.
verified that firmware.bin has changed to FIRMWARE.CUR each time
menuconfig options -- LPC17xx, 100 MHz, and USB selected
both make clean, and make builds have been used, no errors reported at compile
all attempts to flash result in the DFR error 
[image]
 10:46:52
Firmware version: v0.8.0-174-g72161d04-dirty
10:46:52
Connected to host via /tmp/printer @115200bps
10:47:03
Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready

And I've exhausted all available YouTube videos regarding this subject, which is why I'm asking for guidance from y'all. 

Thanks!
V/R,
Jason

Rubber1Duck, 2019-12-15:
Did you follow these instructions ?

"It is necessary to determine the serial port connected to the micro-controller. For micro-controllers that connect via USB, run the following:
ls /dev/serial/by-id/*
It should report something similar to the following:
/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0"

and did you put these output in Printer.cfg?
for me it looks like NO !!!!
Your config:
[mcu]
serial = /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00

on a SKR1.3 it must be somthing like this
[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1768_1810010342941DAF3ABEFC59C02000F5-if00

This part 1810010342941DAF3ABEFC59C02000F5 is depending on the serial id of the USB Chip!

cu
Peter

jasonshrout, 2019-12-15:
Peter,
Ah...  I've tried flashing it that way and definitely missed that in the printer.cfg
This did it though!  Now I have the ability to move the steppers!  

I noticed something odd though.  After running menuconfig and make, the 
ls /dev/serial/by-id/*
identifier kept changing.  Maybe something to do with the clock?  But, all is on it's way to being well now!!

Thank you!!

closed 2019-12-15 by jasonshrout