Klipper3d/klipper #2281: Arduino Due: USB ID it doesn't work but the serial mode does.

robustini, 2019-12-16:
I can't get the printer to work in USB ID but only in serial mode at 250000 baud.
With "make menuconfig" I correctly enter the IDs in the configurator but then the printer does not communicate with Repetier Server.
With "lsusb" i see this:

Bus 001 Device 004: ID 2341:003d Arduino SA

So 0x2341 is the vendor id and 0x003d the product id.
In the printer.cfg:

[mcu]

serial = /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_55736323938351C0A120-if00

With USB ID the result is this one:

9:48:02.932: // Printer is not ready
9:48:02.932: // The klippy host software is attempting to connect. Please
9:48:02.932: // retry in a few moments.
9:48:02.933: !! Printer is not ready

While in serial mode it perfectly communicates:

9:48:04.439: // Klipper state: Ready
9:48:04.482: ok FIRMWARE_VERSION:v0.8.0-193-g380309a0 FIRMWARE_NAME:Klippe

Where am I doing wrong?

robustini, 2019-12-16:
Here the log when it doesn't communicate.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3967174/klippy.log)

robustini, 2019-12-16:
And here the log when it works correctly in serial mode.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3967185/klippy.log)

KevinOConnor, 2019-12-17:
After flashing Klipper with USB support, you must run the `ls /dev/serial/by-id/*` command again to determine the new device name - as described in the Installation document.  (The device name will change after Klipper is flashed to the device.)

-Kevin

robustini, 2019-12-17:
Thanks @KevinOConnor for reply!
Reflashed now but the device name is the same.
After: /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_55736323938351C0A120-if00
Before: /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_55736323938351C0A120-if00

KevinOConnor, 2019-12-17:
If you compile and flash Klipper with USB support, then you must use the other usb plug (the one furthest from the power port).

-Kevin

robustini, 2019-12-17:
You mean the "Native Port" instead "Programming Port" of Arduino Due?

robustini, 2019-12-17:
Ok, with the Native Port work fine, coll thanks!

robustini, 2019-12-21:
I've strange behavior flashing using USB ID with Arduino Due.
After flashing it, everything works perfectly, and the "Native Port" is correctly identified, with "lsusb":

Bus 001 Device 014: ID 2341:003e Arduino SA

Doing an RPI reboot no longer works because it changes:

Bus 001 Device 006: ID 03eb:6124 Atmel Corp. at91sam SAMBA bootloader

To show it again with the correct ID I have to reflash it, and it's terribly boring.
Is it the fault of Raspberry or the firmware that installs Klipper?

robustini, 2019-12-21:
Solved with Arduino Due hardware mod fix.

closed 2019-12-21 by robustini

apphead, 2020-09-01:
Sorry to resurrect this issue, but after struggling with that menuconfig USB for a while, I am wondering what can be gained from using the native port in terms of performance. 
Also, if I understand correctly, using a genuine DUE does not make the flashing procedure any different (apart from the settings of course) or more complicated.