Klipper3d/klipper #2282: Steppers stopping some time into the print

DBa2016, 2019-12-16:
Since few days (without changes made to soft- and hardware), some time into the print, the steppers just stop moving. I see the printer accepting commands (in the Octoprint "Terminal" tab) and reporting back temperature, but none of the steppers seem to move.

This happens sometimes 2 hours after the print start, but sometimes 24 hours after start (in a 50-hour print...).

A "restart" in the terminal window makes the stepeprs move again (but this of course interrupts the print).

Stepper drivers are TMC5160 (all of them), working fine since summer this year.

Log of the last print failure is attached, I issued a "dump_tmc" on all steppers after I noticed the last failure - however I cannot decypher the dmp_tmc output.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3967602/klippy.log)

KevinOConnor, 2019-12-17:
Your driver is reporting an undervoltage error which looks a little suspicious.  The tmc spec describes this as: `Indicates an undervoltage on the charge pump. The driver is disabled during undervoltage. This flag is latched for information.`

Beyond that, I don't see anything that looks abnormal in the log.

-Kevin

DBa2016, 2019-12-17:
Hmmm - wondering how "undervoltage" could happen there... Board failure possible? MKS GEN L on 24V...

DBa2016, 2019-12-17:
More accidentially than really by intent, I discovered the Watterott 5160s have RSense 0.075 Ohm (unlike 0.11 Ohm for other drivers). Changed that and printing again, let's see where we get...

DBa2016, 2019-12-23:
That did not really change the behavior. However, I noticed that after mid-print stop, the drivers show „overload a“ and „overload b“ in the drvstatus flags. Could that be something?

KevinOConnor, 2019-12-23:
Alas, the trinamic drivers are quite intricate.  You'd need to read the trinamic specifications to know the implication of that flag being set.

-Kevin

KevinOConnor, 2020-01-23:
I'm going to close this as it looks like the conversation has concluded.  The trinamic specifications are the best source of information on the behaviour of the trinamic drivers.

-Kevin

closed 2020-01-23 by KevinOConnor