Klipper3d/klipper #2283: Macro repository

wizhippo, 2019-12-16:
I would love to see how others are using macros.  For example right now I'm working on how pause/resume and timeouts interact and would like to see how others may have tackled it.

Is it ok if i ask others to share their examples and have a place to post them?

As for where to place them and additional examples/macros folder would be ok though a bit of a hassle to have to submit PR to add them.

A simpler way would be great though not sure what that would be.

JohnEdwa, 2019-12-18:
Github has wiki support ([the OctoPrint Wiki for example](https://github.com/foosel/OctoPrint/wiki)), which could be a good place to gather them. Wouldn't require permanently adding them to the codebase with a pull request, but would keep them closely tied to the code.

The only problem with it is that there are only two user access controls for it, it's either fully open to anyone with a Github account, or only to people added as "Collaborators" for the project (which then gives them full access to mess with the project code). There is no specific "Wiki write access only" category, which sucks.

KevinOConnor, 2019-12-23:
My high-level feedback is that it sounds interesting.  I don't have any particular ideas on how to implement it however.

FWIW, I've had poor experiences with wikis on several past projects.  I'm not prepared to go down that path again.

-Kevin

wizhippo, 2019-12-23:
The 'how to' is the kicker.

Perhaps gists and let the community know (not sure where, maybe in the gcode template docs) to to prefix the the description with [klipper macro] My cool macro

I'll leave the idea here.  Feel free to close the issue when you feel it has been hanging around long enough.

ld3300, 2019-12-24:
How about putting them in a separate repo?

KevinOConnor, 2020-02-02:
I'm going to close this as it looks like the conversation has concluded.  Alas, I don't have any direct recommendations myself.  If someone has a proposal that they are willing to implement then let us know.

-Kevin

closed 2020-02-02 by KevinOConnor

ld3300, 2020-02-02:
Okay, so I created a repo on my account that all are welcome to use.  I'll put some up there and see if anyone else contributes.  Might need help with organizational plan if anyone is interested in participating. [Klipper-Samples](https://github.com/ld3300/Klipper-Samples)