Klipper3d/klipper #2286: Error thrown when switching extruders with Chimera

jvdkieft, 2019-12-16:
[dual_extusion_calibration_2.gcode.txt](https://github.com/KevinOConnor/klipper/files/3970479/dual_extusion_calibration_2.gcode.txt)
I just put an SKR v1.3 and TH3D chimera clone on my CR-10S. I can get either extruder to print on its own, but when I try and do a dual extrusion print from S3D (or Cura) I'm getting an error when it does a tool change "Internal error on command:" and then G1 for S3D or M104 for Cura. Using the TH3D sample profiles for the dual extruder. Klippy.log and the Gcode attached.

This is from a master pull of Klipper from last night when I rebuilt everything just in case.

 The octoprint.log for the particular section looks like this:

2019-12-14 22:08:33,716 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Internal error on command:"G1"

| Last lines in terminal:

| Send: N3990 G1 X150.003 Y140.810 E-0.9789*72

| Recv: ok

| Send: N3991 G1 X149.023 Y140.762 E-0.9550*79

| Recv: ok

| Recv: ok

| Send: N3992 G1 E-7.1291 F600*19

| Send: N3993 M117 26% L=11/40*111

| Recv: ok

| Send: N3994 T0*13

| Recv: ok

| Recv: // Activating extruder extruder

| Send: N3995 G1 Z2.180 F1002*36

| Recv: ok

| Recv: ok

| Send: N3996 G1 X153.719 Y137.071 F4800*125

| Send: N3997 G1 X154.491 Y137.343 E0.0294 F1232*0

| Recv: ok

| Send: N3998 G1 X154.662 Y137.413 E0.0066*104

| Recv: // Klipper state: Shutdown

| Recv: !! Internal error on command:"G1"

And it repeats the Internal error on command "G1" over and over. This happens on a tool change. 
[dual_extusion_calibration_2.gcode.txt](https://github.com/KevinOConnor/klipper/files/3970481/dual_extusion_calibration_2.gcode.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3970482/klippy.log)

KevinOConnor, 2019-12-17:
There was an error in the ACTIVATE_EXTRUDER, which should be fixed now (commit 46e7027b).

-Kevin

jvdkieft, 2019-12-17:
That worked, now I just have to get the Chimera calibrated!

KevinOConnor, 2019-12-23:
Great.  I'm going to close this issue then as it sounds like the original issue was resolved.

closed 2019-12-23 by KevinOConnor