Klipper3d/klipper #2287: filament_switch_sensor pause_on_runout False restores position when it shouldn't

wizhippo, 2019-12-16:
Any ideas why after `runout_gcode` below for case 1 does not stay at the parked position an returns to the last postin and case 2 does not when a runout happens?

Both call `pause` but case 1 does so manually via macro while case 2 relies on `pause_on_runout: True`

Case: 1
```
[gcode_macro PRINT_PAUSE]
gcode:
  PAUSE

[filament_switch_sensor extruder_filament_runout_sensor]
pause_on_runout: False
runout_gcode:
  PRINT_PAUSE
  PRINT_PARK
  FILAMENT_UNLOAD
switch_pin: ar15
```

Case: 2
```
[filament_switch_sensor extruder_filament_runout_sensor]
runout_gcode:
  PRINT_PARK
  FILAMENT_UNLOAD
switch_pin: ar15
```

Arksine, 2019-12-17:
Even though it appears that the position is being "restored", what is actually happening is that there is a gcode left in the queue that executes after the `runout_gcode` executes.  When `pause_on_runout` is enabled the filament_switch_sensor module immediately notifies Octoprint that it must pause.  Doing this makes sure that Octoprint does not send another gcode after the `runout_gcode` executes.

wizhippo, 2019-12-17:
So the pause_on_runout is amune to this?  Does it flush the queue prior to running the gcode?   Is there anyway to flush that last move when we call pause ourselves in run_out_gcode?  I know in the example above the print_pause only calls pause but it reality it will end up doing more and allows reuse of the macro elsewhere.

Arksine, 2019-12-17:
The purpose of the pause_on_runout option is to properly handle this situation.  It does not flush the gcode queue per se (although Klipper may do this internally), it immediately notifies Octoprint that it must pause, it is not possible to do this from within the `runout_gcode`.  That said, you can still call the macro that contains the PAUSE, Klipper will simply ignore the second pause.

wizhippo, 2019-12-17:
One last question.  Does this then mean that other macros that use `PAUSE` could also have this issue then as octoprint could queue commands in between the `PAUSE` and the rest of the commands in the macro? Is it best then to always run `PAUSE` first then followed by the macro? For example to implement an `M600` you would want to send `PAUSE` first then the `M600` instead of including `PAUSE` in the `M600` macro?

Hywelmartin, 2019-12-17:
My Macros looks like this.. but I print only from SD-card

wizhippo, 2019-12-17:
@Hywelmartin thank you.  I'm going to add a PR that allows to extend the internal PAUSE/RESUME with additional gcode to run.

Arksine, 2019-12-17:
This does not affect pause_resume.  Its unique to the filament sensor because the pause is driven by a printer event.  When PAUSE iis called via gcode Octoprint will stop correctly because it receives the pause command before it receives an acknowledgement from Klipper

wizhippo, 2019-12-18:
I'm finding I have this issue if I use a macro in `runout_gcode`

For example the below exhibits the error and the printer moves back to the workspace after parking. 

``` 
[gcode_macro PRINT_PARK]
gcode:
  SAVE_GCODE_STATE NAME=PRINT_PARK
  G91
  G1 Z{printer["gcode_macro global"].z_lift} F{printer["gcode_macro global"].travel_speed}
  G90
  G1 X{printer["gcode_macro global"].brush_x} Y{printer["gcode_macro global"].brush_y_start} F{printer["gcode_macro global"].travel_speed}
  RESTORE_GCODE_STATE NAME=PRINT_PARK

[filament_switch_sensor extruder_runout_sensor]
runout_gcode:
  PRINT_PARK
switch_pin: ar15
```

Yet if place the gcode directly and call no macro in `runout_gcode` like below I do not have this issue.

For example the below exhibits the error and the printer performs the next gcode in the queue  

``` 
[filament_switch_sensor extruder_runout_sensor]
runout_gcode:
  SAVE_GCODE_STATE NAME=PRINT_PARK
  G91
  G1 Z{printer["gcode_macro global"].z_lift} F{printer["gcode_macro global"].travel_speed}
  G90
  G1 X{printer["gcode_macro global"].brush_x} Y{printer["gcode_macro global"].brush_y_start} F{printer["gcode_macro global"].travel_speed}
  RESTORE_GCODE_STATE NAME=PRINT_PARK
switch_pin: ar15
```

wizhippo, 2019-12-18:
Grr,  Nope it is hit or miss even with no macro involved.

Sometime it stays parked and others it does not.

Is this an octoprint issue as it is slipping in a last instruction between `pause` and `paused`?

Arksine, 2019-12-18:
Do you have "pause_on_runout" enabled?  If not, you must enable it for a runout event to pause correctly. 

This is not an issue with Klipper or Octoprint, it is a result of the limitations of gcode.  This behavior is the reason the "pause_on_runout" option for the filament_switch_sensor exists.

wizhippo, 2019-12-18:
@Arksine the config above I do not set it as it is true by default.  So yes it is enabled.

Arksine, 2019-12-18:
Ok, this is most likely not related to the macro.  Typically this behavior is exhibited when Octoprint "swallows" an acknowledgement sometime prior to the PAUSE.  Its one of the strange things Octoprint does to keep compatibility with firmwares that don't correctly acknowledge all commands.

You can try increasing the "pause_delay" option in the filament switch sensor to attempt to work around this.  It delays adding the "runout_gcode" to the gcode queue in an effort to let Octoprint catch up.

The better option is to attempt to determine what is causing the acknowledgement to be swallowed.  If you have a long running command that triggers Octoprint to timeout and retry it will often swallow the acknowledgement.  If this is happening you can try playing with Octoprint's timeouts to prevent it from happening.

wizhippo, 2019-12-18:
@Arksine I'll keep looking and try extending the pause_delay.

Here is what I see in octo print
```
| Last lines in terminal:
| Send: N58 G1 X216.247 Y24.994 E17.061*88
| Recv: ok
| Send: N59 G1 X216.247 Y22.674 E17.32142*95
| Recv: // action:paused
| Printer signalled that it paused, switching state...
| Changing monitoring state from "Printing" to "Pausing"
| Recv: ok
| Send: N60 G1 X216.287 Y22.117 E17.38411*82
```

I find it odd that octoprint sends N60 even though it is aware it is switching to pausing.

wizhippo, 2019-12-19:
pause_delay: 1 seems to have fixed it.

closed 2019-12-19 by wizhippo