Klipper3d/klipper #2288: 'mcu' shutdown does not return to safe state

BoerTheun, 2019-12-17:
I have just experienced an mcu shutdown during printing. 
BUT the mcu did not turn off any heaters as a result my entire print is melted and I had a lot of smoke when I entered the room.

Why does the MCU do not cut off all heaters when entering shutdown mode?

glowtape, 2019-12-17:
What controller board are you using, and how is your heatbed controlled (24V or mains via SSR)?

BoerTheun, 2019-12-17:
The board given with the velleman k8200 3d printer. My heated bed is controlled via a DC relay with a 24V Siemens PSU.

glowtape, 2019-12-17:
Per log, the MCU shutdown because the ADC went out of range. It either means a thermistor failed or a the maximum temperature of a heater was exceeded. When the MCU shuts down, IIRC all outputs should zero and any connected MOSFETs and switches ought to react accordingly.

You say your print melted, so that means the heatbed didn't turn off? Any idea about the extruder heater? If only the heatbed kept heating, combined with the type of error (temp. exceeded), it suggests that the relay failed and got stuck in ON position.

Do you have any way to check the function of the relay?

glowtape, 2019-12-17:
Took a second look at the log, there's actually temps in there. Heatbed reached at least 144°C and the print head reports up to 458°C (?!?). But the log reports them cooling down before you yanked the power. Maybe the controller crashed in another wild fashion and eventually rebooted.

KevinOConnor, 2019-12-17:
Have you run the temperature and M112 checks described at: https://www.klipper3d.org/Config_checks.html ?  Those steps are intended to verify the micro-controller shutdown logic.

You may also wish to verify that the hardware power relays aren't enabled if their control lines float.  (This is an indication of a wiring fault or a design failure of the control board.)  To test, remove power from the printer, run `sudo service klipper stop`, power up the printer while the host software is not running, and verify that the heaters are not enabled.

-Kevin

BoerTheun, 2019-12-17:
> But the log reports them cooling down before you yanked the power. 

This klipper log is from after my emergency shutdown. The first reported temps are 144°C and 458°C.

I verified M112 again and it works. The power relay is not enabled on a floating control line. It is a Omron MY4IN 24VDC. I believe my printer went into shutdown because my extruder heat limit was 250 and I was printing at 235. A small sudden spike caused the trip. I have increased the limit to 300.

KevinOConnor, 2019-12-23:
The Klipper micro-controller code was designed to transition to a shutdown state should the heater temperature exceed the configured range.  The shutdown state is designed to disable the heaters.  It's possible a hardware fault could have resulted in the failure (some power failures can result in the micro-controller not running at all - even such that the internal "watchdog" events don't occur).  It's also possible there is a severe software fault, but I haven't seen an indication of that.  The M112 test is intended to test those code paths.

FWIW, if my printer had that type of failure, I would be very careful about continuing to use the hardware without tracking down the root cause.

-Kevin

BoerTheun, 2019-12-27:
I am closer to tracking down the problem. I can not do a firmware reset. Klipper fails to recover from executing that command. I have to press the manual reset button on the k8200 board to recover.

What does the firmware_reset do?

```
09:32:03 Klipper state: Disconnect
09:32:07 Failed automated reset of MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Klipper state: Not ready
09:32:07 Failed automated reset of MCU 'mcu'
```

KevinOConnor, 2019-12-29:
By default, the Klipper host software will attempt to reset the micro-controller using the "arduino" reset mechanism (toggling the serial DTR line).  If your board doesn't support that, try adding `restart: command` to the mcu config section (see config/example.cfg for the details).

FYI, I don't see how the host reset mechanism would impact a heater runaway.

-Kevin

KevinOConnor, 2020-01-23:
Okay, I'm going to close this as it looks like the conversation has concluded.  To the best of our knowledge, the micro-controller software will attempt to turn off the heaters in the event of an error.

-Kevin

closed 2020-01-23 by KevinOConnor