Klipper3d/klipper #2292: Worse print quality with M205

Buerk, 2019-12-18:
**First of all thanks for this great Software!**

I've installed Klipper 2 weeks ago for my Ender 3 Pro (no modifications appart from Klipper-FW).
I had some troubles finding a correct pressure_advance setting but meanwhile it's mostly ok.

However I had an odd behaviour yesterday while printing 2 times the same part (same gcode):
The part was sliced with cura 4.4 with all M201..M205 settings commented out so klipper settings are used (other print settings see below).
Printing took place from virtual SD card

The first part was printed without any restrictions for acceleration (part A in the photo).
The second part (part B in the photos) was printed with restricted acceleration by issuing command M204 P500.00 R1000.00 T500.00 in the terminal before printing.

The reason for limiting acceleration was part A had some minor blobs on the outer edge. So the idea was to limit acceleration (or more precise deceleration) to avoid shooting over the edge during deceleration. But the result was even worse quality with more round corners / blobs as you can see on the photos. 

Any idea what could cause this behaviour?? I really thought slower acceleration/deceleration would increase quality by lowering force on the belts and frame.

Thanks in advance

Best regards

Richard.

============================
Cura 4.4
Based on Ender 3 Profile

Modifications:
Speedsettings increased:
Print Speed: 70mm/s
Wall Spped:  50mm/s
Top/Bottom Speed: 50mm/s
(Travel Speed: 175 mm/s - default)
Initial Layer Speed: 30 mm/s

Retraction: 4 mm
Retraction Window: 6

All M201..M205 commands in gcode start disabled so the klipper config is used.

===========================

Klipper Settings:
Based on Ender 3 Profile

Modifications:
max_accel 2000
square_corner_velocity: 5.0 (added explicitly)

pressure_advance: 0.88

[virtual_sdcard]
path: ~/.octoprint/uploads/

[klippy.log](https://github.com/KevinOConnor/klipper/files/3978144/klippy.log)
[image]
[image]
[image]
[image]

JohnEdwa, 2019-12-18:
Reducing acceleration means the nozzle would have a longer time to ooze (or release pressure, more correctly) during the corner move. I also remember reading from somewhere that the Klipper motion planner doesn't like having the acceleration changed during printing, though I can't validate that.

@KevinOConnor 
The [Gcodes.md](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#g-code-commands) file says that M204 only supports the "S" parameter, but [the code also "supports" P and T](https://github.com/KevinOConnor/klipper/blob/master/klippy/toolhead.py#L555) though in a non-standard way (P is supposed to be for printing move acceleration and T is for travel move acceleration, the code selects the smaller for everything), but where does that "S" come from?   
[It's not a standard parameter for M204 on the RepRap wiki](https://reprap.org/wiki/G-code#M204:_Set_default_acceleration) and is only briefly mentioned as a "other firmwares" parameter.

robschwieb, 2019-12-18:
> Reducing acceleration means the nozzle would have a longer time to ooze (or release pressure, more correctly) during the corner move. I also remember reading from somewhere that the Klipper motion planner doesn't like having the acceleration changed during printing, though I can't validate that.
> 

Any chance someone else could comments on acceleration changes mid print that causes issues with the motion planner? I'd really like to read more on that.

Buerk, 2019-12-18:
Thank you for all your qualified feedback! I really aprecciate this!
Just to make sure, there's no misunderstanding:
I've issued the command M204 P500.00 R1000.00 T500.00 right before the second print (print B) was started. So there was no acceleration change during printing.
By the way: the command above is default from Cura 4.4with Ender3 Profile as part of start gcode along some other M20x commands which I've removed.

@JohnEdwa:
> Reducing acceleration means the nozzle would have a longer time to ooze (or release pressure, more correctly) during the corner move. I also remember reading from somewhere that the Klipper motion planner doesn't like having the acceleration changed during printing, though I can't validate 
that.

That could be a possible reason, however I thought that the corner speed is controlled by the parameter _square_corner_velocity_ and the deceleration takes place before. I also have pressure_advance active with a relative high value which should avoid oozing.
The idea of reducing acceleration and deceleration was, that the nozzle does not shot over the target by reducing forces on the belt and the frame. During first print the printer already shaking a bit, still yielding better results.

KevinOConnor, 2019-12-18:
> Any chance someone else could comment on acceleration changes mid print that causes issues with the motion planner?

I'm not aware of any problem reports related to changing of acceleration during a print.

> The Gcodes.md file says that M204 only supports the "S" parameter, but the code also "supports" P and T

I've updated the documentation (commit b921b8a1).

> though in a non-standard way

Haha, "standard g-code".  Very funny.

> It's not a standard parameter for M204 on the RepRap wiki and is only briefly mentioned as a "other firmwares" parameter.

Slic3r emits S (when acceleration is enabled) and Cura used to emit it.  Cura now emits P and T (set to the same value) when acceleration is enabled, which is why Klipper now accepts that form.

-Kevin

KevinOConnor, 2019-12-18:
> I had some troubles finding a correct pressure_advance setting but meanwhile it's mostly ok.

What troubles did you have?

> The reason for limiting acceleration was part A had some minor blobs on the outer edge. So the idea was to limit acceleration (or more precise deceleration) to avoid shooting over the edge during [...]

I can't really tell from the pictures.  However, your description sounds like issues due to extruder pressure.

What you're describing "avoid shooting over the edge during deceleration" would help if you are experiencing "ringing".  Going faster through the corners (via a high acceleration and/or high square_corner_velocity) can make ringing worse but reduce the impact of extruder pressure.  The oppose - going slower through corners, magnifies the impact of extruder pressure, but can reduce the effect of ringing.  Unfortunately, ringing and pressure problems can result in similar looking corners.

-Kevin

JohnEdwa, 2019-12-18:
>Haha, "standard g-code". Very funny.

Fine, "de-facto standard", because I don't think following [DIN 66025](http://www.helmancnc.com/din-66025-nc-programming-codes/) would get us very far. 

---

[Which does mean this is Klipper extended G-codes in a nutshell.](https://imgs.xkcd.com/comics/standards.png)

Hywelmartin, 2019-12-18:
@JohnEdwa  The S was removed from Marlin Mar 11-2015
I use REPRAP/SPRINTER flavor when I slice.. so that I can alter accelerations per extrusion typ... no problems...

Buerk, 2019-12-19:
@KevinOConnor:

> >I had some troubles finding a correct pressure_advance setting but meanwhile it's mostly ok.

>What troubles did you have?

Tuning pressure_advance to higher values directly made a change from blobing at corners into under extrusion. I didn't find a setting where the corners got a round shape.

For investigating ringing, I've printed a calibration cube:
Printing from virtual SD-card, both time the same print-file / gcode file.
Print A with my settings as above (acceleration by Klipper config 2000mm/s²)
Print B with acceleration limited to 500mm/s² by gcode M204 P500.00 R1000.00 T500.00 right before the printing of print B was started.

[image]
[image]
[image]

There is a little ringing/ghosting (do both have the same meaning???)  on the Y-axis with the higher acceleration (print A).
However the overall quality is again a bit worse with lower acceleration (print B): rounder corners and some small gaps in the finish of the top-layer on the z-axis.

Thanks for your help!

Buerk, 2019-12-19:
I've printed some pressure calibrate squares.
It turns out, that pressure advance doesn't eliminate the blobs at the edges.
Settings used: 
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500
Print A: SET_PRESSURE_ADVANCE advance=0.0
Print B: SET_PRESSURE_ADVANCE advance=0.88
Print C: SET_PRESSURE_ADVANCE advance=1.0
Layer height 0,3mm, other settings in Cura as above.

From print A to print B there's nearly no improvement...
I did also start a print with presure advance set to 1,2 which still showed the blobs in printing direction. I had to cancel the print since the extruder started to make really strange noise (sounded like the feeder slipping across the filament).

For me it looks like the nozzle somehow shoots behind the target position - maybe due to the mass of the printhead - since the blobs are prominent in the printing direction where deceleration occurs (I've drawn the outer perimeter printing direction into picture of print B). But I don't undestand why this is more prominent with lower acceleration. This is somehow strange since it still occurs even with high pressure_advance settings. Another argument against pressure_advance setting is, if these blobs where originated from oozing they should be more uniform at the edges - or am I wrong?

@KevinOConnor 
For better understanding:
Does Klipper precalculate the cornering or junction speed based on the square_corner_velocity and decelaration stops some distance before the "real" edge - or to be more precise deceleration is moved into an circular movement with a constant centripetal acceleration? As I understand, the goal is to achieve a constant centripetal acceleration. Is this correct?

[image]
[image]
[image]

KevinOConnor, 2019-12-23:
I'm not sure what the state of this issue is.

I can confirm that your description and pictures continue to follow the common theme: high accel results in ringing; low accel results in bulging at the corners due to extruder pressure.

The Klipper kinematics are documented at https://www.klipper3d.org/Kinematics.html .  Klipper never alters the tool path to a "circular movement".

Your pressure advance of ~1.000 is very high.  It may be possible to lower that value by tuning the e-steps, tuning the printing temperature, or by altering the bowden tube (eg, by tightening them, using better tubing, or shortening the tube) - there is some links at: https://www.klipper3d.org/Pressure_Advance.html .  It's also possible a different filament may yield better results.  If you do retest pressure advance, I recommend you use the "tuning tower" method described in the latest guide (see link above).

Finally, if you can't tune pressure advance to a lower value, I suspect you'll have to accept the trade off between ringing and extruder pressure.

-Kevin

Buerk, 2019-12-26:
I've changed to standard (slow) Cura Settings for Ender 3 for feedrate (50 and 25mm/s) and acceleration (500mm/s²).
With this setup I could change pressure_advance down to 0 and get acceptable results though the printing speed is very low and the benefits of Klipper are gone.
Higher feedrates directly lead to the bulging edges nearly regardless which setting for pressure_advance I'm choosing...

closed 2019-12-26 by Buerk