Klipper3d/klipper #2294: "Commercial" use of the Klipper Logo (not really an issue)

pkElectronics, 2019-12-18:
Hi there,

im currently working on a new kind of 3D printer control board and would like to use the Klipper logo on my website to show compatibility of the board with klipper. Are there any requirements to do so? A preview of the site can be found here: www.progress-3d.de

During the course of development I´ll contribute to the project by porting Klipper to the STM32MP1 which will run Klipper on it´s real time core and an instance of OctoPrint on it´s dual core ARM cpu.

Best Regards
Pascal

ProfDrYoMan, 2019-12-19:
Interesting board. What's the release schedule plan? Roughly.

JohnEdwa, 2019-12-19:
Everything in the Klipper project, the logo included, [is released under GPL v3.0](https://github.com/KevinOConnor/klipper/blob/master/COPYING) which means commercial use is perfectly fine, but you have to add a section on your site with the required license/copyright information, and if you want to go all 100% CYA legalese you need to figure out how the part of GPL that requires you to release with the same license follows when using a GPL licenced image in your project. (Which it probably doesn't, but IANAL).

I'm sure there are plenty of guides on how you legally use GPL licensed content around if you search for it.

pkElectronics, 2019-12-19:
@JohnEdwa thanks for pointing me in the right direction

@ProfDrYoMan I´m afraid this will take some time, maybe in summer 2020. Have to find an investor to help me fund the production, certification and global distribution

closed 2019-12-19 by pkElectronics