Klipper3d/klipper #2297: Failure in serial connect

YaronSoffer, 2019-12-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3984812/klippy.log)

There is something wrong with the serial connection after updating the flash and klipper version to the current day. please advise

BlackStump, 2019-12-19:
good idea to check here before/after update
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md
>20191126: The USB names have changed on micro-controllers with native USB support. They now >use a unique chip id by default (where available). If an "mcu" config section uses a "serial" setting >that starts with "/dev/serial/by-id/" then it may be necessary to update the config. Run "ls >/dev/serial/by-id/*" in an ssh terminal to determine the new id.

KevinOConnor, 2019-12-20:
The log indicates that the host was unable to connect to the micro-controller.  What micro-controller is this, and what settings did you use in "make menuconfig"?  (Either take a screenshot of "make menuconfig" or attach the `~/klipper/.config` file here.)

-Kevin

YaronSoffer, 2019-12-21:
[config.zip](https://github.com/KevinOConnor/klipper/files/3991360/config.zip)
I am adding the config - it is a Duet3d (wifi).

As for the name on the serial , they look the same.

orangepi@OrangePi:~$ ls -l /dev/serial/by-id/usb-03eb_6124-if00
lrwxrwxrwx 1 root root 13 Dec 21 10:34 **/dev/serial/by-id/usb-03eb_6124-if00** -> ../../ttyACM0
orangepi@OrangePi:~$ cat ~/printer.cfg | grep serial
serial: **/dev/serial/by-id/usb-03eb_6124-if00**

KevinOConnor, 2019-12-23:
It does not look like Klipper was successfully flashed to the given device.  (It it was successfully flashed, the USB id would have reported a Klipper specific name.)

Nothing was changed recently with the sam4e8e support and I'm not aware of recent trouble reports on that board.

-Kevin

YaronSoffer, 2019-12-26:
Hi, 
After reflashing everything back to normal.
Thanks a lot for the tip and the help.
by the way I have to do on orangepi  "sudo make flash ...." otherwise it will not work for me.

closed 2019-12-26 by YaronSoffer