Klipper3d/klipper #2298: "Aus3D Rumba32" board not recognized by RPI after flashing with Klipper

robustini, 2019-12-19:
I did a lot of tests but the board is seen by Raspberry only in DFU mode (STM32 Bootloader) during flash, after it it is not detected even with "lsusb".
I tried also with the related ST Tool to flash the klipper.bin file, it is written but the same problem, after reboot is not recognized.
The chip is STM32F446:

https://github.com/Aus3D/RUMBA32

robustini, 2019-12-19:
Soon even MKS will sell its version:

https://github.com/makerbase-mks/MKS-RUMBA32

lorenzing, 2019-12-20:
Hi, I have the same problem with my Rumba32

KevinOConnor, 2019-12-20:
The "make menuconfig" program produces a file called `~/klipper/.config`.  Can you attach that file here?  (Or, alternatively, produce a screen shot showing the settings that you used in "make menuconfig".)

-Kevin

robustini, 2019-12-20:
I use this settings:

[image]

KevinOConnor, 2019-12-20:
The schematic from the website you linked to above indicates the board is using a 12MHz crystal.  So, at a minimum you would need to set that and flash the board again.

I'm surprised you have disabled USB in the config.  If you are trying to communicate over the USB port on the board then that option must be enabled.

-Kevin

robustini, 2019-12-20:
I try, thanks!

robustini, 2019-12-22:
Work fine.

closed 2019-12-22 by robustini