Klipper3d/klipper #2299: [Feature Request] Simultaneous homing

mtw3d, 2019-12-19:
Is it possible to make the X & Y axis home simultaneously? It would save time on large format printers, and shouldn't introduce any significant inaccuracy. 

Simultaneous homing of Z would also probably be a nice option, however it would need to be configurable since it wouldn't work with a probe. I would propose something like:

```
[simultaneous_homing]
axes: XYZ
```

I don't believe any other configuration options would be required.

My understanding is that deltas already home simultaneously, so I don't believe this should take much effort to implement.

mtw3d, 2019-12-19:
Shep suggested in Discord that for complex situations like a 5 axis machine, being able to specify the homing order would also be nice, so I suggested something like a comma separated list such as:

`axis_homing_order: xy, z, ab`

Any axes that are grouped together would be homed simultaneously, in the order they appear in the list.

This should only be the behavior of a plain `G28`. Calling something like `G28 X` should still behave just as it does now.

vladbabii, 2019-12-19:
i would sugest just making something like this work
```
G28 XY Z UV AB
```
with a default in config

But! I think there is one issue with homing both at the same time
* on a bedslinger, it's fine
* but on a corexy you will hit at least one endstop at an angle instead of straight and i think that's a good way to destroy it faster

- Also some optical endstops don't like if you come from an angle
- also, on a bedsling, home xy and Z at the same time can lead to some nice tears from broken probes, bent bltouch tips

If this is implemented it should come with a warning - use at your own risk

mtw3d, 2019-12-19:
Yeah, it absolutely must be optional, and it should not be the default. There are definitely places where it wouldn't be a good idea. But for many circumstances, it will save seconds. Definitely not a "must have" feature, but it seems like it should be easy to implement.

mtw3d, 2019-12-19:
> i would sugest just making something like this work

>G28 XY Z UV AB

> with a default in config

Actually, as I think about it,  depending on how it was implemented this could be dangerous. Imagine a user typed:

`G28 XYZ`

If the system just homed any axes that were grouped simultaneously, that could be lead to unexpected results. 

As long as G28 with options is treated as defined in `axis_homing_order` that would be great, but to avoid over-complicating the implementation just homing the axes one at a time when options are passed to G28 would be fine.

JohnEdwa, 2019-12-20:
In Marlin, these are configuration options called `QUICK_HOME` and `HOME_Y_BEFORE_X`, so I'll use those terms. 

Gcode commands don't care about the order of parameters and adding that to G28 only would be rather confusing for users. Also, if you want to home just X and Y without Z, the command is `G28 X Y` or `G28 Y X`. `XY` or `YX` are invalid parameters and so technically they could be added as the ordered ones, though if something like this would be implemented it should definitely be an extended gcode command.

Currently, the code parses the `G28` parameters, then stick them into an array where X = 0, Y = 1 and Z = 2, and runs the homing code in order. Implementing `HOME_Y_BEFORE_X` should be rather easy to do, `QUICK_HOME` would require a little more work.

mtw3d, 2019-12-20:
Whether it is valid or not isn't what really matters, it is whether a user might think it was valid and type it unintentionally. 

I don't have a Marlin printer handy, but I just checked with a friend who still runs Marlin to confirm that G28 XY is a valid command  that homes X then Y. So anyone coming from Marlin might think G28 XYZ was valid and type it, not understanding the consequences. Given that it is a potentially dangerous command, I don't think handling it that way in the terminal makes sense. 

I think making the standard G28 command support simultaneous homing has a significant benefit, but I don't think the benefit is so big that we need to make it that customizable in the terminal unless we can do it in a way that is safe. The syntax I suggested above is for the config file, not the terminal, so presumably you will be taking the time to think about what you are doing. 

As for [quick_home] vs. [simultaneous_homing], I only prefer the latter because it is completely unambiguous. Harder to spell, but easier to understand without reading the docs. But either work for me.

[home_y_before_x] becomes completely redundant when you can specify the order completely.

KevinOConnor, 2019-12-23:
FWIW, my high-level feedback is that implementing "faster homing" is not interesting to me.  The goal of the homing code is to get highly accurate results - in particular, we really want the homing code to repeatably measure the Z steppers to within an accuracy of around 20 microns.  That's a difficult task and there is a lot of code complexity around implementing that on different types of printers.  In contrast, a savings of 20 seconds during homing results in a time savings of ~0.07% of an 8 hour print.  So, I'd be leery about adding additional code complexity to that area of the code.

-Kevin

JohnEdwa, 2019-12-23:
For a Cartesian, the only difference (and how Marlin does it IIRC) is that you start the homing sequence with "move until you hit an endstop" for both axes at once, and once one of them hits it does the bump back and homes properly. I can't actually even remember how it behaves if Y hits before X, I think it just waits for the X homing to finish and then does the Y if `HOME_Y_BEFORE_X` isn't enabled.

And IIRC it's not applicable for a Delta or CoreXY printers at all, as their homing works differently.

KevinOConnor, 2020-01-23:
Okay, I'm going to close this as it looks like the conversation has concluded.

-Kevin

closed 2020-01-23 by KevinOConnor

maruhe, 2021-05-22:
I just switched from Marlin and I have a CoreXY
and there some features I currently miss in Klipper, one is Marlin's QUICK_HOME

for me it looks like this thread got a little confused
for a printer like mine it makes sense to home X & Y together - I'd never Home Z in the same step as my Z-endstop is my Probe-sensor..
but on bigger beds (400mm in my case) it's nice to save time ;-)

marlin did a diagonal (45°) XY move toward the endstops and didn't mind about wich one hit's first @JohnEdwa 
after both axes hit the sensor the second (slow) probe was done one by one

and @accurate homing, on X and Y I have homing speed on 100 for first (and will tune up) and 10 for second probe and I hope just the second value is used ;-)