Klipper3d/klipper #2300: Custom thermistor using Beta value

Fabrizio3, 2019-12-19:
this is a quick and dumb problem, i have a new NTC thermistor that is a bit different from generic, the manufacturer sent me the temp tables along with a B value, its not specified or in an example config, so, how do you define the Beta type and correctly set the values? been trying for half an hour and it either doesn't read or doesn't understand it, when i try to define it in the config it doesn't load. Thanks! :)

jakep82, 2019-12-19:
The standard supported thermistors are listed [here](https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L154).

If it's not one of those, you'll need to define a custom thermistor as described [here](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L818)

robustini, 2019-12-20:
It would be interesting to be able to insert a custom Beta value instead of having to customize the thermistor.
For example for "NTC 100K beta 3950" being able to shift that 3950 to a custom value.
I do the same with the MK4duo firmware and it is very convenient.

Fabrizio3, 2019-12-20:
does anyone have an example on how to define a new thermistor altogether? cause mine has a beta value of 4287K or something like that and different resistance values, I seen some guy do the same for a slice engineering thermistor but I don't  know how he did it (not too tech savvy with code, Im good at hardware)

KevinOConnor, 2019-12-20:
The config/generic-printrboard-G2.cfg file has an example of a custom thermistor.  The relevant section is:

```
[thermistor G2]
temperature1: 20
resistance1: 140000
temperature2: 195
resistance2: 593
temperature3: 255
resistance3: 189
```

If you don't have three temperatures from a temperature chart, and instead just have one temperature/resistance and beta, the config would look something like:

```
[thermistor my_thermistor]
temperature1: 25
resistance1: 100000
beta: 4287
```

-Kevin

Fabrizio3, 2019-12-20:
can't I just add it on the default sensors in klipper/klippy/extras/thermistor.py and define resistance temp and beta like the rest?

jakep82, 2019-12-20:
Every time you update Klipper, any changes you make to thermistor.py will be lost.  If you use that exact same data to define it in your printer.cfg, it will be persistent through updates.

Fabrizio3, 2019-12-20:
oh, I didn't know resistance had to be set to 100000, that seemes to have fixed it
I was using a resistance from the thermistor's table, 126.7276 ohms at 20°

robustini, 2019-12-20:
> 
> 
> oh, I didn't know resistance had to be set to 100000, that seemes to have fixed it
> I was using a resistance from the thermistor's table, 126.7276 ohms at 20°

Write here your thermistor table data.

Fabrizio3, 2019-12-20:
it states in the datasheet 25° 100k, I just didn't know I had to use that, I probably did something dumb with the 126k value and it didn't work, it's all fixed and working, thanks guys?

closed 2019-12-20 by Fabrizio3

rafaljot, 2019-12-26:
Is theree any option similar to Marlin?
It is convenient to be able to check the temperature and ADC during tuning thermistor table.

from Marlin configuration_adv.h :
`
// Show Temperature ADC value
// Enable for M105 to include ADC values read from temperature sensors.
//#define SHOW_TEMP_ADC_VALUES
`

KevinOConnor, 2019-12-26:
See the QUERY_ADC command described at: https://www.klipper3d.org/G-Codes.html

-Kevin

rafaljot, 2019-12-27:
After some code research I found out how to do that, thx Kevin for showing the direction.:
1) heat up slowly and read responses of this command
query_adc name="extruder" pullup=4700

example for 200'C 
Warning:  ADC object "extruder" has value 0.292186 (timestamp 2318.190)
Warning:  resistance **1940.159** (with 4700 pullup)

Measure temperature using external thermometer and collect data.  

2) Write section like this (this is for RAMPS-FD and 100K beta 3950, there is odd thermistor section in RAMPS-FD ):
[adc_temperature my_thermistor]
temperature1: 10
resistance1: 5000000
temperature2: 15
resistance2: 3300000
temperature3: 20
resistance3: 1500000
temperature4: 25
resistance4: 945000
temperature5: 30
resistance5: 480000
temperature6: 40
resistance6: 300000
temperature7: 50
resistance7: 175000
temperature8: 60
resistance8: 105000
temperature9: 70
resistance9: 66000
temperature10: 80
resistance10: 44000
temperature11: 90
resistance11: 33000
temperature12: 100
resistance12: 23000
temperature13: 110
resistance13: 17000
temperature14: 160
resistance14: 4020
temperature15: 200
resistance15: 1940
temperature16: 210
resistance16: 1300
temperature17: 220
resistance17: 1015
temperature18: 240
resistance18: 840
temperature19: 250
resistance19: 712
temperature20: 260
resistance20: 600
temperature21: 280
resistance21: 525
temperature22: 300
resistance22: 0

finally use it for extruder, heater_bed, etc:
[extruder]
(...)
sensor_type: my_thermistor
pullup_resistor: 4700

Rubber1Duck, 2019-12-27:
Hi,

why not simply (as kevin wrote)

> .......................................................
>
> If you don't have three temperatures from a temperature chart, and instead just have one temperature/resistance and beta, the config would [...]

?????????????????????????????????
For your example 100K beta 3950 thermistor

[thermistor my_thermistor]
temperature1: 25
resistance1: 100000
beta: 3950
?????????????????????????????????

rafaljot, 2019-12-27:
Because it is RAMPS-FD. There is no simple pullup/inline resistors and beta curve doesn't work.
[image]

KevinOConnor, 2019-12-29:
FYI, my read of that schematic is that it just requires `pullup: 1000` in the heater section.

-Kevin

rafaljot, 2019-12-29:
I doesn't work. I tried to use 1K pullup it shows apx -50'C at room temperature. 
In addition, this solution is unstable at low temperatures but it seems to work better over 100'C then simple thermistor + two resistors.
That's why I have to tune it for 0-50'C. I do the same in Marlin.
I thing there is sth odd with vref