Klipper3d/klipper #2301: adding to virtual_sd.py the possibility to fetch gcode files recursively

manu7irl, 2019-12-20:
Hi I am adding files in my virtual_sd by folder created under DWC2 UI, and in order to print directly from the LCD, I need the virtual_sd.py to fetch files recursively as DWC2 create a subfolder named GCODES and from there I want to be able to print them from my lcd unit.

Also I made an auto mount function for hot swapable thumb drive that mounts itself in a subfolder under the virtual_sdcard folder, so with this change I will be able to print without the need of a physical nor wireless connection to my printer to get my prints done!
thanks @KevinOConnor for your great work.

My python skills sucks, but I think that a simple change in this funtion will help to get the job done,
this is the ooriginal function:

` def get_file_list(self):
        dname = self.sdcard_dirname
        try:
            filenames = os.listdir(self.sdcard_dirname)
            return [(fname, os.path.getsize(os.path.join(dname, fname)))
                    for fname in sorted(filenames, key=str.lower)
                    if not fname.startswith('.')
                    and os.path.isfile((os.path.join(dname, fname)))]
        except:
            logging.exception("virtual_sdcard get_file_list")
            raise self.gcode.error("Unable to get file list")`

and this is my proposal:

`def get_file_list(self):
        dname = self.sdcard_dirname
        try:
            for root, dirs, files in os.walk(self.sdcard_dirname):
                for file in files:
                if file.endswith(".gcode"):
                    filenames = (os.path.join(root, file))
            return [(fname, os.path.getsize(os.path.join(dname, fname)))
                    for fname in sorted(filenames, key=str.lower)
                    if not fname.startswith('.')
                    and os.path.isfile((os.path.join(dname, fname)))]
        except:
            logging.exception("virtual_sdcard get_file_list")
            raise self.gcode.error("Unable to get file list")`

KevinOConnor, 2019-12-23:
This look like a dup of #1476 

-Kevin

manu7irl, 2019-12-25:
Yes this looks like in your reference someone asked for about the same thing... I will post there.

closed 2020-01-03 by KevinOConnor