Klipper3d/klipper #2302: Dry Run mode, test print without extrusion and temp, possible?

robustini, 2019-12-20:
I am trying to perform XYZ (Delta) handling tests and I wish it would not extrude.
I did slicing by putting all the temperatures at zero but when the printing starts then Klipper blocks it with the obvious error:

// Extrude below minimum temp
// See the 'min_extrude_temp' config option for details
!! Extrude below minimum temp

Is there a way in Klipper fer let him ignore extrusion and temperatures without going to upset the printer.cfg?

tonn0297, 2019-12-20:
octoprint have a plugin that works https://plugins.octoprint.org/plugins/DryRun/

robustini, 2019-12-21:
I know, but I use Repetier Server, and it doesn't have this plugin.
It would be correct that Klipper could do it without plugin.

Hywelmartin, 2019-12-21:
slice without extruder..

robustini, 2019-12-21:
> 
> 
> slice without extruder..

No, it doesn't work, already tried.
Only the DryMode plugin for Octoprint work fine.

Spacemarine2018, 2019-12-22:
Workaround: Set min_extrude_temp = 0 and remove the connector of the extruder motor.

robustini, 2019-12-22:
> 
> 
> Workaround: Set min_extrude_temp = 0 and remove the connector of the extruder motor.

This is a better workaround... :-)

[image]

closed 2019-12-27 by klipper-gitissuebot