Klipper3d/klipper #2303: analog14

Maik0605, 2019-12-20:
Hi.

I have flashed Klipper compltete new to make an Update.
Yesterday all was Ok
Now i have this Problem

 Unknown value 'analog14' in enumeration 'pin'
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
Klipper state: Not ready

What can i do ?

Maik0605, 2019-12-20:
Klippy Log is there
[klippy.log](https://github.com/KevinOConnor/klipper/files/3990486/klippy.log)

KevinOConnor, 2019-12-23:
If you want to use the "arduino" pin aliases, then you'll need to add `pin_map: arduino` to the mcu section of the config file.  (See config/example.cfg for the details.)

-Kevin

KevinOConnor, 2020-01-03:
I'm going to close this as I think the original question has been answered.

-Kevin

closed 2020-01-03 by KevinOConnor