Klipper3d/klipper #2304: Ender 3 Pro with Bltouch MCU 'mcu' shutdown: Can't assign oid

droidmahn, 2019-12-21:
So I just copied over the config I use on 2 other ender 3 pros that print fine (adjusting the mcu path ofc) but it errors out with "MCU 'mcu' shutdown: Can't assign oid" 
This is a fresh install of Klipper, octoklipper, octoprint, and a fresh flash. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3991136/klippy.log)

KevinOConnor, 2019-12-23:
Your micro-controller is rebooting when the host attempts to configure it.  Make sure the micro-controller actually is an atmega1284p (and not some other type of AVR chip).  You can also try recompiling and flashing the chip (`make distclean ; make menuconfig ; make flash`).

-Kevin

KevinOConnor, 2020-01-11:
Okay, I'm going to close this as it looks like the conversation has concluded.

-Kevin

closed 2020-01-11 by KevinOConnor