Klipper3d/klipper #2305: Sensorless homing with SKR 1.3 & TMC2209 v3 (FYSTEC)

KoffeinKaio, 2019-12-21:
First of all - klippy.log: https://gist.github.com/KoffeinKaio/d105cb50064a79651d6243d7b69e5da7

Hey, 
I just recieved the TMC2209 V3 ( https://wiki.fysetc.com/images/TMC2209.jpg , https://wiki.fysetc.com/Silent2209/ ) and cannot get Sensorless homing to work with klipper.
Altough I had a problem getting them to work at all, the issue https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/59#issuecomment-515607375 worked.

I can measure a direct connection between diag and signal pin of x endstop - but it always says "endstop still triggered".

I tried !P1.29, ^P1.29, ^!P1.29 and (which should be the right one if I understand the wiki) P1.29

Any guesses?

deudtens163, 2019-12-21:
Hi,
i had the same problem today and i had to disable the second homing movement as said in the sensorless_homing document.
In your config file, for each axis section using  sensorless homing you have to add :
homing_retract_dist: 0
I hope can be usefull for you

KevinOConnor, 2019-12-30:
The best info we have on sensorless homing is in the docs at: https://www.klipper3d.org/Sensorless_Homing.html .   It does sound like `homing_retract_dist: 0` is needed.

-Kevin

KoffeinKaio, 2020-01-13:
Sorry for the long delay - but I had to take a break from printing.

The Sensorless Homing still wont work, even with homing_retract_dist.

`
[tmc2209 stepper_x]
uart_pin: P1.17
microsteps: 4
run_current: 0.500
hold_current: 0.300
#stealthchop_threshold: 100
diag_pin: ^!P1.29
driver_SGTHRS: 1
`
With diag_pin: ^!P1.29 and diag_pin: !P1.29 you can hear the motor hum for a split second - but it doesnt move.

boardthatpowder, 2020-01-16:
Here's my config for SKR 1.3 with TMC2209's, with senseless homing working:  https://github.com/boardthatpowder/klipper-firmware/blob/master/printer.cfg.

My differences include:
- your currents are way lower than mine (`0.8` & `0.5` vs `0.5` & `0.3`)
- my diag_pin is `P1.29` vs your `^!P1.29`
- with `driver_SGTHRS` of `1` you have sensorless homing practically disabled.  255 is maximum sensitivity.  I run mine at 120 but this needs to be tweaked to your setup.  Start at about `160` and fine tune from there

And as mentioned above, you need `homing_retract_dist: 0` must be set for the stepper.

KevinOConnor, 2020-02-14:
Okay, I'm going to close this as it looks like the conversation has concluded.  As best we know, the sensorless homing code in Klipper works as designed - it is known that tuning the tmc drivers to reliably get a signal can be a difficult process.  The best info we have on that is in https://www.klipper3d.org/Sensorless_Homing.html and in the trinamic documentation.

-Kevin

closed 2020-02-14 by KevinOConnor

pusewicz, 2020-05-14:
@boardthatpowder Any particular reason why you have disabled the sensorless homing in your current config? I can't tell from the commit. I was about to enable sensorless on my SKR1.3 + 2209 
 (currently X & Y does not move, so I guess I need to lower the sensitivity).

boardthatpowder, 2020-05-14:
@pusewicz i disabled sensorless homing and went back to physical endstop switches as I found I was continuously tweaking the thresholds to get them to work Effectively. Sometimes the sensorless homing would kick in too soon and fail prints, whereas others it would not kick in at all.

pusewicz, 2020-05-14:
@boardthatpowder Gotcha. Thanks! Will not look into sensor-less homing then and stick to physical switches.