Klipper3d/klipper #2306: Anet A8 Klipper 0.8 "Printer is halted"  Problem

ohmegacorp, 2019-12-21:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3991247/klippy.log)

[printer.txt](https://github.com/KevinOConnor/klipper/files/3991248/printer.txt)

Hi;
After I've updated klipper to 0.8 I'm facing "printer is halted" problem. I have successfully flash my anet a8 board . When I start octoprint an try to connect, it is connecting succesfully 
[image]

but when I try to control my printer it gives error message 

[image]

Do you have an idea about that?

BlackStump, 2019-12-21:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md
>20190606: The "samples", "samples_result", and "sample_retract_dist" config options have been >moved to the "probe" config section. These options are no longer supported in the "delta_calibrate", >"bed_tilt", "bed_mesh", "screws_tilt_adjust", "z_tilt", or "quad_gantry_level" config sections.

ohmegacorp, 2019-12-21:
Problem solved 👍🏻

closed 2019-12-21 by ohmegacorp

krpepe, 2019-12-21:
@ohmegacorpYou need to look at the new example.cfg file to see what changed. Sample under your config file is no longer valid and needs to be corrected. Sent via the Samsung Galaxy S8 Active, an AT&T 5G Evolution capable smartphone

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/KevinOConnor/klipper/issues/2306?email_source=notifications\u0026email_token=AGMAWGB5FK47QAUFUJ6CLBDQZXPHZA5CNFSM4J6GILC2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4ICDV3BQ",
"url": "https://github.com/KevinOConnor/klipper/issues/2306?email_source=notifications\u0026email_token=AGMAWGB5FK47QAUFUJ6CLBDQZXPHZA5CNFSM4J6GILC2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4ICDV3BQ",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]