Klipper3d/klipper #2307: gcode macro variable_MYVARIABLE not working

theopensourcerer, 2019-12-21:
I have a simple macro for moving my nozzle to a cleaning brush:

[gcode_macro clean_nozzle]
gcode:
  G28
  G0 Z1 F1000
  {% for wipe in range(10) %}
    {% for coords in [(285,-21), (240,-21)] %}
      G0 X{coords[0]} Y{coords[1] - 0.6 * wipe} F6000
    {% endfor %}
  {% endfor %}
  G28

I wanted to use a couple of variables at the top, so according to the example here: https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L408

I changed my macro thus:
[gcode_macro clean_nozzle]
variable_XMIN: 240
variable_XMAX: 285
gcode:
  G28
  G0 Z1 F1000
  {% for wipe in range(10) %}
    {% for coords in [(XMAX,-21), (XMIN,-21)] %}
      G0 X{coords[0]} Y{coords[1] - 0.6 * wipe} F6000
    {% endfor %}
  {% endfor %}
  G28

This didn't work. When klipper tried to execute the macro I got:

Recv: !! Unable to parse move 'G0 X Y-21.0 F6000'

If, however, I replace:

 variable_XMIN: 240 
 variable_XMAX: 285

with:

 default_parameter_XMIN: 240
 default_parameter_XMAX: 285

then XMIN & XMAX are parsed correctly and the macro works.

theopensourcerer, 2019-12-21:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3991792/klippy.log.zip)

JohnEdwa, 2019-12-22:
Variables seem to need to be in lowercase letters. 
```
[gcode_macro asd]
variable_xmin: 240
variable_XMAX: 285
gcode:
  M118 xmin= {xmin}
  M118 XMAX= {XMAX}
```

> Send: asd
> Recv: echo: xmin= 240
> Recv: echo: XMAX=

theopensourcerer, 2019-12-22:
@JohnEdwa Thanks. You are correct. it appears that uppercase characters are not working at all. (I tried "variable_xMax" for example and this was not parsed either.

I am not sure if this is a bug or a feature but if the latter then presumably this condition should be included in the documentation.

KevinOConnor, 2019-12-23:
It would appear that the python config reading tools will change the config settings to lower case.  It looks like this would be something to document.

-Kevin

KevinOConnor, 2020-01-11:
FYI, the docs have now been updated for this (commit 6972b630).

Thanks,
-Kevin

closed 2020-01-23 by KevinOConnor