Klipper3d/klipper #2311: Implementing Purge Bucket

dpackham, 2019-12-22:
Working on implementing something like this below.  it uses a purge bucket vs a purge block to save waste.  I need to write a M211 Gcode macro and figure out where in the printer.cfg file the "EXTRUDER START" and "END" codes and I need ideas.

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3993135/printer.cfg.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3993131/klippy.log)

I requires an "extruder change" gcode to move the nozzle over the bucket and purges some filament before resuming the print.

EXTRUDER START G-CODE
; Start Extruder 1
M211 S0 ; Turn off soft stops
G1 X-9.5 F6000 ; Go to X=-9.5 at 100 mm/sec
G92 E0 ; Reset extrusion distance on new tool
G1 E50 F90 ; Extrude 50 mm of filament at 1.5 mm/s on new tool
M211 S1
G92 E0 ; Set current extruder position to 0mm
G0 F10000 ; Increase feed rate before next travel.

EXTRUDER END G-CODE
; End Extruder 1
G92 E0 ; Reset extrusion distance
G1 E-5 F5000; Retract 5 mm
G91 ; Switch from absolute positioning to relative positioning
G1 Z1 F1000 ; Lift nozzle by 1 mm in relative positioning mode
G90 ; Return to absolute positioning
G92 E0 ; Reset extrusion distance on old tool

[https://www.thingiverse.com/thing:3395883](url)

JohnEdwa, 2019-12-23:
As it says right there on the page:

>Settings > Printer > Manage Printer > Machine Settings > Extruder (1,2) Tabs. Paste the following in the extruder start/end G-code for EACH extruder.

"M211 - Software Endstops" probably isn't something Klipper has, but it doesn't really need them either. All it does is allow you to move beyond your min and max positions set for the steppers in your printer.cfg, but you can just set them to the actual max and min positions in the first place. I have my `[stepper_x] position_max` set to 253, as that is a few mm before it physically hits my belt tensioner bracket, even though my bed is 235mm.

dpackham, 2019-12-23:
If I was using cura that would work.  What I was more asking was is there a macro like "start_print" that I could put this gcode into and use that for example "change_extruder". Then I just put that into cura or s3d

JohnEdwa, 2019-12-23:
The `START_PRINT` and `END_PRINT` are just standard Klipper gcode macros, you can create ones for extruder changes and call them from Cura instead of putting the whole Gcode snipped there.

dpackham, 2019-12-24:
ohh.  didn't know I could create new ones.  I will get them working and share them here

KevinOConnor, 2020-01-11:
Any further updates on this?

-Kevin

dpackham, 2020-01-13:
Yes.  I have tested a side purge bucket and am waiting for the parts for a push to purge bucket.  I will have you a testable config in a few days.

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like the conversation has concluded.  Feel free to update this ticket when ready.

-Kevin

closed 2020-02-14 by KevinOConnor