Klipper3d/klipper #2312: MCU 'mcu' shutdown: Timer too close

metalinspired, 2019-12-23:
Let me start by saying I find Klipper amazing despite the issue I'm having.
So far I googled every answer to my questions/problem but this one yielded no result that helped me.
As you can see from the title it's, yet another, _MCU 'mcu' shutdown: Timer too close_ issue.
I've attached my Anet A8 to a Odroid C1+ board using direct serial connection, meaning it's not using USB cable, Klipper and Octoprint are installed each under separate user and, from what I concluded looking at system logs, there is nothing that even remotely suggests what could cause the issue in question.
In [klippy.log](https://github.com/KevinOConnor/klipper/files/3995322/klippy.log) you can see it occurring multiple times at random intervals.
Most of the time the only way I can get communication back up is to hold reset on Anet board for couple of seconds before attempting host reset.

KevinOConnor, 2019-12-23:
Your log indicates the serial communication is not stable.  Specifically, both failures show the comms being halted for 600+ms (the host reads no data for an extended period, and then reads dozens of messages that were generated in the past).  Eventually this unstable communication results in the micro-controller triggering a safety check.

It's also possible that severe cpu starvation on the host machine could be causing the issue, but that seems less likely.

-Kevin

metalinspired, 2019-12-24:
I sincerely doubt it has anything to do with host CPU as it has more than enough power.
On the other hand I see no reason for unstable communication.
I've switched to USB connection and it seemed to work flawlessly but on the last layer of the same print I lost connection once again but this time with different message.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3998446/klippy.log)

KevinOConnor, 2019-12-24:
That log indicates a "lost communication with mcu" error - the best advice we have for that is at: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

It's also worth noting that ModemManager (or similar OS packages) have been known to mess with connections in odd ways.  If that software (or similar) is running, it might also explain this and previous errors.

-Kevin

metalinspired, 2019-12-29:
I was about to call it a win after disabling ModemManager and successful 2 and 5 hours prints but today it struck again.
However, this time, I got an actual error (hub 1-1:1.0: port 4 disabled by hub (EMI?), re-enabling...) in system logs
This pretty much shows that klipper is not the culprit here.

closed 2019-12-29 by metalinspired