Klipper3d/klipper #2314: [Feature request] Don't require each of X, Y, Z and E to be present.

Spacemarine2018, 2019-12-23:
I'd like to use a separate board, separate RPI and separate Klipper installation to drive only manual steppers, as this works very well in building mechatronic systems that help my 3D-printer (Filament changer, build-plate remover, etc.)

The problem is that each klipper installation requires X,Y,Z,E to be configured and bound to real existing pins. This reduces the number of available pins to use for manual steppers. On My SKR PRO with 6 stepper-drivers, I can therefore use only 2 manual steppers.

It would be desireable if klipper only stops with an error if a move of an axis that is not present, is requested instead of stopping right after loading the printer.cfg.

KevinOConnor, 2019-12-23:
If your device does not have a traditional toolhead, then you can use `kinematics: none` in the printer config section.  The "none" kinematics does not define any of x, y, z steppers.  It is not necessary to define an extruder on any kinematics.

-Kevin

Spacemarine2018, 2019-12-24:
Wow, I was not aware of that, despite reading the docs intensively!

This solves my problem, thanks again!

closed 2019-12-24 by Spacemarine2018