Klipper3d/klipper #2315: Cetus3d Setup

Scottyprogrammer, 2019-12-23:
So I've been trying to setup my cetus with the tinyfab cpu and ive got it running, but not printing., The axes seems to have the wrong steps despite using the ones directly from the cetus firmware. I can't imagine they are so far off because I keep adjusting the z steps to no avail. I thought maybe it could be the driver current because it was set in the firmware but I'm not sure how to include this in the printer config. I've attached the klippy log and the config I was using on the cetus when it was running on the tinyfab cpu.
[config.txt](https://github.com/KevinOConnor/klipper/files/3996317/config.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3996308/klippy.log)

KevinOConnor, 2019-12-23:
FYI, there were several past reports on the Cetus3d printers.  Have you read through those issues?  https://github.com/KevinOConnor/klipper/issues?q=is%3Aissue+cetus+is%3Aclosed

-Kevin

Scottyprogrammer, 2019-12-24:
I thought I had read through them all. Not sure how I overlooked the most recent issue with what sounds like the same problem. I'm I'll test it out when I'm back by my printer. Thanks.

Scottyprogrammer, 2019-12-24:
dharvey242 was correct in his issue. I had to change the pulse duration to 3 before compiling the firmware. All working great. Thank you.

closed 2019-12-24 by Scottyprogrammer