Klipper3d/klipper #2316: Pressure advance test document requests a print speed which is above standard hotends capability

ProfDrYoMan, 2019-12-24:
Hi,

The pressure advance test document requests a print speed which is above standard hotends capability.

See: https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md

> Use a slicer to generate g-code for the large hollow square found in docs/prints/square_tower.stl. Use a high speed (eg, 100mm/s), zero infill, and a coarse layer height (the layer height should be around 75% of the nozzle diameter).

Assuming 0.4 mm nozzle width I expect every slicer to generate 0.45 mm wide traces.
75% of nozzle width leads to 0.3 mm layer height.

=> 100 mm/s * 0.45 mm * 0.3 mm = 13.5 mm^3/s

E3D v6 is known to handle 8 mm^3/s on PLA pretty save. For me working with 10 mm^3/s seems to be still ok, But I can see the difference on bigger top infills.

KevinOConnor, 2019-12-26:
The values in Pressure_Advance.md are given as examples - starting points for a user to begin with.  There is no one speed/setting that will work with all printers.

-Kevin

ProfDrYoMan, 2019-12-26:
I am not sure if this is well known. I guess it might screw up the PA values. If someone is putting the max flow into his slicer, the generated c-code will be slower anyhow.

Maybe some small information for the user?

slavedas, 2020-01-15:
I regularly run faster than that on an MK8 hotend without issue at higher temperatures.  I have no signs of underextrusion and good dimensionality.  Is this a hard and fast rule or just a guideline people like E3D put out to CYA (not impugning them, they make fantastic products, just saying they probably need to make sure they can point to something that says "See you aren't running it like we said")?

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like the conversation has concluded.  The goal of the PA docs is to give an example speed near the high end so as emphasize nozzle pressure.  In general, I get good feedback on 100mm/s.  It wont work for all printers, but I'm not sure any value will work well for all printers.  So I'm inclined to leave it like it is.

Thanks for the feedback.
-Kevin

closed 2020-02-14 by KevinOConnor