Klipper3d/klipper #2318: Extrude only move too long (100.000mm vs 100.000mm)

ld3300, 2019-12-24:
Having a weird issue with a macro I'm experimenting with to swap filaments.  The macro doesn't cause problems when called manually, only during a print.  The print is failing with Extrude only move too long (100.000mm vs 100.000mm) as seen on line 41618 of the log.  Both of the extruders have a max_extrude_only_distance of 100.0 (line 39091).  The command that seems to be breaking it is g-code macro "LOAD_EXTR" on line 38858.

When I get a T1 command I am checking my variables to see if T0 is actively in the hot end. If so I run "PARK_EXTR" (Line 38845) which seems to be accomplishing successfully.  After the tool switch it attempts to run "LOAD_EXTR" for T1 and fails with the above error.  

Is it possible this is some sort of floating point error?  My inclination is just to increase my max_extrude_only_distance to 100.1 to avoid the issue, but thought it was worth investigating.

Ignore the earlier error where I attempt to extrude 130mm, that was clearly my bad.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3999049/klippy.log)

robustini, 2019-12-25:
You must increase your max_extrude_only_distance.

KevinOConnor, 2019-12-26:
It does sound like a floating point rounding issue - the simplest solution is to configure a slightly larger maximum.

If you think there is a separate issue, you should run M112 immediately after the event so that the log has more information - see: https://www.klipper3d.org/Contact.html

-Kevin

ld3300, 2019-12-27:
Thanks Kevin, I set the new max to 100.1 to avoid the issues.  When I get bored I will try to reproduce it with a 112 to see what there is to see.

KevinOConnor, 2020-01-23:
Okay, it looks like we can close this then.

-Kevin

closed 2020-01-23 by KevinOConnor