Klipper3d/klipper #2319: Reset on non TMC-XX Controllers for enstop-phase detection?

dothebart, 2019-12-25:
I've read through the [Calibrating endstop phases documentation](https://www.klipper3d.org/Endstop_Phase.html) and it occurs to me that this may be one of the problems I've got to make my delta printer u(anycubic kossel) with the mks-smini Smoothieboard-clone working reliably - it will not precisely match the bed from a run into its endstops. After a run of `DELTA_CALIBRATE` with a BL-Touch it will work better, but I've learned that this is intended to be ran only once since it will override other settings from a run with the delta calibration rod.

The guide mentions, that the TMC Reset-circuit is vital to the process, and that elseways a complicated procedure of de-powering the controllerboard is neccessary. The smini as well as one generation of smoothie board use the Heroic [HR4982](https://www.datasheet4u.com/datasheet-pdf/HEROIC/HR4982/pdf.php?id=1256520) which also has a reset-line. It seems as if the schematics have disappeared from the smoothieboard webpage - thus I can't tell whether - and where - they are connected to the MCU. Would this reset-cirquit qualify for this procedure? 

Could a different homing procedure improve the quality? Like:
- home once fast to travel the distance
- step off the switches a bit 
- home each tower on its own with slower speed (maybe several times?)

KevinOConnor, 2019-12-26:
One could theoretically use the stepper driver reset line to ensure the micro-controller and stepper driver stay in sync.  However, I'm not aware of any board that has the stepper driver reset lines routed to the micro-controller.  Should one manually route the reset line, it would require software development work.

Using a slow second_homing_speed configuration may improve the accuracy of the endstop switches.  That's totally separate from endstop phases however.

-Kevin

dothebart, 2019-12-29:
It seems that this fixes my problems, and I get reliable print results. Thank you very much.

closed 2019-12-29 by dothebart