Klipper3d/klipper #2320: src/avr/Makefile:34: flash] Error 1  on Ender 3

mickwo, 2019-12-25:
Hi when i try to flash i have this Error

**I try with this**
ls /dev/serial/by-id/*        
/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
ls /dev/serial/by-path/*                                 
/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.1:1.0-port0

sudo service klipper stop

make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 ( First test)

make flash FLASH_DEVICE=/dev/ttyUSB0 (Second test)
make flash FLASH_DEVICE=/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.1:1.0-port0 (3th test)

**Same Result 3 tiemes**

  Flashing out/klipper.elf.hex to /dev/ttyUSB0 via avrdude
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x46
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x69
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6e

avrdude done.  Thank you.

make: *** [src/avr/Makefile:34: flash] Error 1

The octoprint is Disconnected all time

Any suggestions are welcome

jakep82, 2019-12-25:
https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-2018.cfg#L5

mickwo, 2019-12-26:
> https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-2018.cfg#L5

Thanks , but I forgot to say that I had already tried that alternative, the result is the same ,  only difference is no the line : make: *** [src/avr/Makefile:34: flash] Error 1

mickwo, 2019-12-26:
> Hi @mickwo,
>
> It did not look like there was a Klipper log file attached to this ticket. The log file has been engineered to answer common questions the Klipper developers have about the software and [...]

[klippy.log](https://github.com/KevinOConnor/klipper/files/4001514/klippy.log)

ld3300, 2019-12-27:
Will you try following the update process to make sure you are on the latest master and try again?

drphil3d, 2019-12-27:
You need to power the board. It’s just talking to the usb to serial adapter
but not the mcu because it’s not powered.
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

Buerk, 2019-12-27:
Do you have a bootloader installed on your board (arduino, sanguino,... depending on your board)?
A lot of boards (Creality, Tronxy, ...) don't have the bootloader preinstalled.
Serial flashing only works with bootloader on the avr.

mickwo, 2020-01-04:
Sorry for the late , i found the solution , the answer of Buerk was the solution  I don't have the bootloader preinstalled.

closed 2020-01-04 by mickwo