Klipper3d/klipper #2321: Creality BLtouch V3.0 ANTLABS issue #1938 & #2170

witte1000, 2019-12-26:
Hi Guy's,

Merry Christmas !

I've tried to install Klipper on my Ender 5 with a Creality3D V1.1.4 motherboard and original Creality BLtouch V3.0 ANTLABS.

When running BLTOUCH_DEBUG COMMAND=pin_down and BLTOUCH_DEBUG COMMAND=pin_up everything works like expected, also like describe in the documentation the red and blue led works like expected, but when running G28 and trigger it with my finger the bed keeps raising, it doesn't stop, so the BLtouch works but it don't trigger the Z stop signal.

Next I tried BLTouch 3.1 #1938 and BLTouch improvements #2170 , but then I get 
"Recv: // Internal error during connect: MCU_stepper instance has no attribute 'add_to_endstop'"
 so now I'm stuck ...

Any suggestions are very appreciated !!!
Ps: I'm also struggling with the runout sensor ...

Thanks in advance !

Kind Regards,
[klippy.log](https://github.com/KevinOConnor/klipper/files/4002189/klippy.log)

ld3300, 2019-12-27:
Hi @witte1000 ,

For the runout sensor try setting the pin's pullup and wire the sensor to pull the line to ground when pressed.

Can you also do a clean install without the BLTOUCH python file addon and run G28 then do M112 right after you push the pin and the bed fails to stop?  Then upload the log.

You can also try the following to see if the board is actually seeing a change state from the sensor pin.  Temporarily assign an unused pin the the BLTOUCH sensor (Where you currently have PC4)

Add a button to your config that will print output of the sensor pin. This will send it to the display and to the terminal page on octoprint.
```
[gcode_button bl_test_button]
pin: ^PC4
press_gcode:
    M117 Press Detected
    RESPOND TYPE=error MSG="Press Detected"
release_gcode:
    M117 Release Detected
    RESPOND TYPE=error MSG="Release Detected"
```

witte1000, 2019-12-27:
Hello ld3300,

Thank you very much for your kind feedback !
I appreciate it very much !

_"For the runout sensor try setting the pin's pullup and wire the sensor to pull the line to ground when pressed."_

I don't know if I really understand exactly what you mean, but in Marlin bugfixes 2.x.x the runout sensor is working great like expected, I do use pin 29 and the GND pin on the creality motherboard and use the Z endstop for runout sensor, because I don't needed anymore instead using the BLtouch which is connected to a original creality BLtouch pin 27 and set the following in the config file

[filament_switch_sensor my_sensor] replace it now with [filament_switch_sensor 0]
pause_on_runout: True
switch_pin: PA2

Normally I (thought) should see with the QUERY_ENDSTOPS command the state results of all sensors but I only see xyz, maybe I'm wrong in this ... 
I didn't had the chance to effective do a print with klipper yet because of the BLtouch issue

_"Can you also do a clean install without the BLTOUCH python file addon and run G28 then do M112 right after you push the pin and the bed fails to stop? Then upload the log."_

Sure no problem in the attachment you will find the new klippy.log file,I run it twice, I do have to reflash all the time because I cannot stop printing ...

in the log I do see 
Failed to verify BLTouch probe is raised; retrying.
Failed to verify BLTouch probe is raised; retrying.
Failed to home z: Timeout during endstop homing

_"You can also try the following to see if the board is actually seeing a change state from the sensor pin. Temporarily assign an unused pin the the BLTOUCH sensor (Where you currently have PC4)"_

I'm sorry I'm from Belgium and English is not my mother language and also a newbie in 3D printing and don't understand this maybe I'm stupid  ...
What do you exactly mean with _"assign an unused pin the the BLTOUCH sensor (Where you currently have PC4)"_
do you mean I disconnected the BLtouch and create the virtual button ? 
the unused pin = virtually ?

The BLtouch is connected to a original creality pin 27 in Marlin bugfixes 2.x.x the sensor is working great like expected, I don't see why I should test the sensor pin if it's working perfectly in Marlin where it give signal ? 

Can you explain me exactly what and how then I surely will test it  ...

Thank you !

Kind regards,
[klippy.log](https://github.com/KevinOConnor/klipper/files/4006234/klippy.log)

ld3300, 2019-12-28:
Thank you for doing that.  I see where it is not sensing the BLtouch pin, but am not adept enough yet to see anything in the log to point us in the right direction.  Hopefully someone else will be able to chime in as well.

My suggestion for the "Unused Pin" was to put something in the BLtouch spot a a placeholder, a pin that's not being used like I think PA0 may work.

Then create a macro like I described and using the `BLTOUCH_DEBUG COMMAND=pin_down` and  `BLTOUCH_DEBUG COMMAND=pin_up` and see what is printed in the terminal and you can help determine if the switching signal is making it to the signal pin.  It could help us determine where the issue may be.  

For the filament switch sensor "0" may be a difficult name for klipper to handle.  Try using something like:

`[filament_switch_sensor sensor0]`

Also doing an internal pullup may help the switch be more accurate if your switch connections are from signal pin to ground:

`switch_pin = ^PA2`

You should try the following command when the sensor is pushed and when it is open:

`QUERY_FILAMENT_SENSOR SENSOR=sensor0`

FanDjango, 2019-12-30:
Have you seen this? [link](https://www.antclabs.com/creality3d-board?fbclid=IwAR3QdEJKMqMwluMPNzmAhOiQ5pq6zgGtXtF8PIXY_Ekb4hyMKL4NLKkJjRE). Probably worth reading.

I know that does not necessarily correspond to your printer, but:

There was no end to the troubles with CREALITY printers and the V3.0 BLTOUCH. It took ages for someone to realize that there was something on the breakout board (of that printer type). 

I don't know your printer, but the BLTOUCH V3.0 has very weak drive to the upper rail on it's 10ms output pulse, so in Marlin (**until _finally_ the information about the cable harness** was discovered) we recommended a). a pull-up of 1K to 5V b). Using SW mode to prolong the 10ms pulse (in case it was being mangled by a capacitor) c). Setting 5V mode in the probe (which didn't help all that much because that was weak as well.

**It should probably work without all that**. It was all needless panic because of failures with one specific printer manufacturer - CREALITY - leading to all sorts of fixes, new versions and conjecture.

In case all else fails and you want to experiment:

Using SW mode and/or using 5V mode: For that you need my modified fork/branch [link](https://github.com/FanDjango/klipper/tree/BLTOUCH-HIGH-SPEED), stock klipper won't give you that right now. 

And success is not guaranteed either, **until you follow the wiring of the trigger signal and the schematics to make sure all is clean** for the signal to reach the controller's pin.

Let me know if I can be of further help and happy new year!

witte1000, 2019-12-31:
Ok Thanks guy's for all your excellent information, So in short they messed up with the V3.0 
It's sad to see there is no real collaboration which holds back people to go in 3d my opinion ...
I gonna check it out and try to do some test like you recommend, I need some time for it specially with the holiday's day's right now ... 

What would be the best I can do for in the future having no issues with the BLtouch or incompatible hardware ?
- Buying another one v3.1  ...
- Buying a sensor (which is not so accurate as the BLTouch)
- Buying BigTreeTech SKR Pro (STM32F407)
- Buying another motherboard Looking to printhat https://phat.wrecklab.com/
- ...

I bought a ender 5 only because it's FLOSS and like the qube it's very stabel, and would like to go for Klipper so everything is compatible also a lot easier then marlin and flashing all the time not to mention the rest of Klipper features, but without ABL which is a must I can not print it's not working for me.

Thank you all guy's for your all your new information gonna check it out and come back to here in a couple of day's.

Happy New Year to everyone and God bless you my friends !!! 
Thanks !!!

KevinOConnor, 2020-01-23:
Any further updates on this ticket?  I'm not sure what the current state is.

-Kevin

witte1000, 2020-01-28:
Hi Guy's,

I'm sorry, I'm in a switch of my job and running out of time right now, hopefully next week when everything is settle, I do have more time again for further testing.

Thanks for understanding.
Kind Regards,

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like the conversation has concluded.  Feel free to update when you have further info.

-Kevin

closed 2020-02-14 by KevinOConnor

witte1000, 2020-04-03:
Hello Guy's,

Today I tested again the klipper firmware with BLTOUCH but sadly without succes,

Everything works like describe even 

"Then while gently pushing the pin up slightly with the nail of your finger run QUERY_PROBE again. Verify the command reports “probe: TRIGGERED”

Send: QUERY_PROBE
Recv: // probe: open

Send: QUERY_PROBE
Recv: // probe: TRIGGERED

is reported correct TRIGGERED, but when running a g28 nothing happens, the bed doesn't stop ???
Failed to verify BLTouch probe is raised; retrying.
Failed to verify BLTouch probe is raised; retrying.
Failed to home z: Timeout during homing

even when playing with the setting false true 5V all doesn't matter ...

Why when all the test are triggering correct ?

In the attachments you find the klippy.log 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4427716/klippy.log)

Thank you very much !
Kind regards,

witte1000, 2020-04-03:
In the attachments the klippy.log with all the 3 test
[klippy.log](https://github.com/KevinOConnor/klipper/files/4427860/klippy.log)

Kind regards,

FanDjango, 2020-04-03:
@witte1000 I have some suggestions:
remove `pin_move_time` to use the default.
are you sure about your z_offset?
can you do a g28 and stop the downward movement with your fingernail when the probe is about halfway down, is that possible or does that also not stop the downward movement. <--- this step is missing to check before you progress to a "full-g28"

KevinOConnor, 2020-04-03:
Unfortunately, the log appears truncated and is missing critical information.  Please be sure to attach the full log file.  (If you did attach the full log file, then something went very wrong - try a `sudo service klipper restart` and attach the log after that.)

-Kevin

witte1000, 2020-04-03:
ok strange new try
[klippy.log](https://github.com/KevinOConnor/klipper/files/4427945/klippy.log)

FanDjango, 2020-04-03:
I something in that log:
```
Filament Sensor 0: runout event detected, Time 7900.60
action:paused
Stats 7901.0: gcodein=19 mcu: mcu_awake=0.009 mcu_task_avg=0.000096 mcu_task_stddev=0.000084 bytes_write=4780 bytes_read=5150 bytes_retransmit=27 bytes_invalid=0 send_seq=244 receive_seq=244 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000109 heater_bed: target=0 temp=22.3 pwm=0.000 print_time=13.111 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.3 pwm=0.000 sysload=0.36 cputime=145.212 memavail=696616
Stats 7902.0: gcodein=19 mcu: mcu_awake=0.009 mcu_task_avg=0.000096 mcu_task_stddev=0.000084 bytes_write=5033 bytes_read=5378 bytes_retransmit=27 bytes_invalid=0 send_seq=266 receive_seq=266 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000109 heater_bed: target=0 temp=22.4 pwm=0.000 print_time=14.111 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.5 pwm=0.000 sysload=0.36 cputime=145.256 memavail=696616
Stats 7903.0: gcodein=19 mcu: mcu_awake=0.009 mcu_task_avg=0.000096 mcu_task_stddev=0.000084 bytes_write=5286 bytes_read=5592 bytes_retransmit=27 bytes_invalid=0 send_seq=288 receive_seq=288 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000104 heater_bed: target=0 temp=22.4 pwm=0.000 print_time=15.111 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.3 pwm=0.000 sysload=0.36 cputime=145.304 memavail=696616
Stats 7904.0: gcodein=19 mcu: mcu_awake=0.021 mcu_task_avg=0.000104 mcu_task_stddev=0.000055 bytes_write=5602 bytes_read=5849 bytes_retransmit=27 bytes_invalid=0 send_seq=309 receive_seq=309 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000104 heater_bed: target=0 temp=22.4 pwm=0.000 print_time=16.093 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.5 pwm=0.000 sysload=0.36 cputime=145.345 memavail=696616
Stats 7905.0: gcodein=19 mcu: mcu_awake=0.021 mcu_task_avg=0.000104 mcu_task_stddev=0.000055 bytes_write=5905 bytes_read=6092 bytes_retransmit=27 bytes_invalid=0 send_seq=332 receive_seq=332 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000103 heater_bed: target=0 temp=22.4 pwm=0.000 print_time=17.118 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.3 pwm=0.000 sysload=0.36 cputime=145.408 memavail=696616
Stats 7906.0: gcodein=19 mcu: mcu_awake=0.021 mcu_task_avg=0.000104 mcu_task_stddev=0.000055 bytes_write=6158 bytes_read=6306 bytes_retransmit=27 bytes_invalid=0 send_seq=354 receive_seq=354 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000111 heater_bed: target=0 temp=22.4 pwm=0.000 print_time=18.118 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.4 pwm=0.000 sysload=0.33 cputime=145.447 memavail=696616
Stats 7907.0: gcodein=19 mcu: mcu_awake=0.021 mcu_task_avg=0.000104 mcu_task_stddev=0.000055 bytes_write=6411 bytes_read=6534 bytes_retransmit=27 bytes_invalid=0 send_seq=376 receive_seq=376 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000111 heater_bed: target=0 temp=22.4 pwm=0.000 print_time=19.118 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.5 pwm=0.000 sysload=0.33 cputime=145.493 memavail=696616
Stats 7908.0: gcodein=19 mcu: mcu_awake=0.021 mcu_task_avg=0.000104 mcu_task_stddev=0.000055 bytes_write=6664 bytes_read=6762 bytes_retransmit=27 bytes_invalid=0 send_seq=398 receive_seq=398 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000111 heater_bed: target=0 temp=22.4 pwm=0.000 print_time=20.118 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.3 pwm=0.000 sysload=0.33 cputime=145.530 memavail=696616
Stats 7909.0: gcodein=19 mcu: mcu_awake=0.022 mcu_task_avg=0.000108 mcu_task_stddev=0.000063 bytes_write=6985 bytes_read=6991 bytes_retransmit=27 bytes_invalid=0 send_seq=419 receive_seq=418 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000112 heater_bed: target=0 temp=22.3 pwm=0.000 print_time=21.137 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.5 pwm=0.000 sysload=0.33 cputime=145.577 memavail=696616
Stats 7910.0: gcodein=19 mcu: mcu_awake=0.022 mcu_task_avg=0.000108 mcu_task_stddev=0.000063 bytes_write=7487 bytes_read=7189 bytes_retransmit=27 bytes_invalid=0 send_seq=432 receive_seq=432 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=16000112 heater_bed: target=0 temp=22.4 pwm=0.000 print_time=24.071 buffer_time=2.131 print_stall=0 extruder: target=0 temp=23.4 pwm=0.000 sysload=0.33 cputime=145.617 memavail=696616
Stats 7911.0: gcodein=19 mcu: mcu_awake=0.022 mcu_task_avg=0.000108 mcu_task_stddev=0.000063 bytes_write=7529 bytes_read=7313 bytes_retransmit=27 bytes_invalid=0 send_seq=435 receive_seq=435 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=8 freq=16000118 heater_bed: target=0 temp=22.3 pwm=0.000 print_time=24.072 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.4 pwm=0.000 sysload=0.31 cputime=145.652 memavail=696616
Stats 7912.0: gcodein=19 mcu: mcu_awake=0.022 mcu_task_avg=0.000108 mcu_task_stddev=0.000063 bytes_write=7578 bytes_read=7428 bytes_retransmit=27 bytes_invalid=0 send_seq=439 receive_seq=439 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000121 heater_bed: target=0 temp=22.4 pwm=0.000 prin [... cut ...] rget=0 temp=22.4 pwm=0.000 print_time=36.122 buffer_time=0.000 print_stall=0 extruder: target=0 temp=23.4 pwm=0.000 sysload=0.26 cputime=146.206 memavail=696648
Transition to shutdown state: Shutdown due to M112 command
Dumping gcode input 5 blocks
Read 7887.532123: 'M105\n'
Read 7890.802978: 'M105\n'
Read 7895.805934: 'M105\n'
Read 7897.744181: 'G28\n'
Read 7924.925471: 'M112\nN3 M112*34\n'
```
Could you try to disable the filament sensor by whatever means possible? Or get it to work correctly?

witte1000, 2020-04-03:
@FanDjango 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4427974/klippy.log)

the same

FanDjango, 2020-04-03:
Huh? It's still in there:
```
[filament_switch_sensor 0]
pause_on_runout = True
switch_pin = PA2
```
For testing, get rid of it

witte1000, 2020-04-03:
yes is the old one, now the new one
[klippy.log](https://github.com/KevinOConnor/klipper/files/4428018/klippy.log)

FanDjango, 2020-04-03:
@witte1000 
It will make no big difference but you can set both
```
pin_up_reports_not_triggered = False
pin_up_touch_mode_reports_triggered = False
```
to true.

For the
```
set_output_mode = 5V
```
that's ok for the Creality 1.1.4 and the BLTouch V3.0

@KevinOConnor If the probe shows a trigger in his tests, these use touch_mode and the trigger signal is a steady high - it is seen ok. But the V3.0 produces a 10ms weak pulse on a normal trigger mode "trigger". That pulse is also mangled a bit by the board - the rise time is not so great and this it might appear shorter than 10ms on the input pin. Could this cause a problem? On my tests, my V3.0 on my MKS SBase had no such problem, but I will retest that tomorrow and attach a scope to see the signal, even if it works here for me. 

@witte1000 I suppose triggering the probe with your finger while it is going down also doesn't stop the movement, right? Try to manually get the nozzle up to 100 before doing the g28 and use your finger.

KevinOConnor, 2020-04-03:
So, to summarize:
1. You have a genuine BL-Touch v3.0
2. The pin_up/pin_down control pin test works as expected.
3. The pin_down/touch_mode/QUERY_PROBE sensor pin test works as expected.
4. The G28 homing to finger nail test fails - the Z movement does not stop.
Is this correct?

My initial reaction is that this sounds like the "capacitor issue" described at the end of docs/BLTouch.md .  How is the BLTouch wired to the printer board?

-Kevin

witte1000, 2020-04-03:
@KevinOConnor 

    You have a genuine BL-Touch v3.0
Should be it's original from Creality like I see all the pictures it's correct 
    The pin_up/pin_down control pin test works as expected.
yes correct works like expected 
    The pin_down/touch_mode/QUERY_PROBE sensor pin test works as expected.
yes correct
    The G28 homing to finger nail test fails - the Z movement does not stop.
yes correct, my bed doesn't stop moving up to the nozzle

FanDjango, 2020-04-03:
@KevinOConnor can we give him a one-or-two liner to activate SW mode after the deploy?

witte1000, 2020-04-03:
like I mentioned before,

I don't know if I really understand exactly what you mean, but in Marlin bugfixes 2.x.x the runout sensor is working great like expected, I do use pin 29 and the GND pin on the creality motherboard and use the Z endstop for runout sensor, because I don't needed anymore instead using the BLtouch which is connected to a original creality BLtouch pin 27 and set the following in the config file

I will see i've i can find the youtube movie

witte1000, 2020-04-03:
here it is https://www.youtube.com/watch?v=gwHpXaj_6xE

FanDjango, 2020-04-03:
@witte1000 You can put the runout sensor back in later. I was worried about it. Right now, let's concentrate on the BLTouch. So the BLTouch is on the Creality Board 1.1.4 pin called pin 27.

Board looks like this?
[image]
Which one did you use for the BLTouch?

witte1000, 2020-04-03:
[image]

witte1000, 2020-04-03:
[image]

witte1000, 2020-04-03:
this last photo is the bltouch
the first you see the blue and white cabel is for the filament sensor

witte1000, 2020-04-03:
[image]

witte1000, 2020-04-03:
[image]

FanDjango, 2020-04-03:
The bltouch is the red plug?

witte1000, 2020-04-03:
red blue and yellow = bltouch

FanDjango, 2020-04-03:
The BLTouch signal is that red plug:
[image]
????

witte1000, 2020-04-03:
it's a seperate card 
[image]

yes the red plug is also BLtouch

the z cabel with white and blue is filament sensor 
[image]

witte1000, 2020-04-03:
[image]

FanDjango, 2020-04-03:
We are talking about the SIGNAL pin of the BLTOUCH. Black and White. 2 poles. See these ones:
[image]
Forget the filament sensor for now. It has nothing to do with this.

FanDjango, 2020-04-03:
So the BLTOUCH trigger signal black and white are connected at the Z-Stop pin, which carries a 4.7uF capacitor. Duh.

witte1000, 2020-04-03:
yes correct white and black / yellow red blue / red plug on the board => BLtouch
on the board white and blue cabel = filament sensor

FanDjango, 2020-04-03:
OK, I will test a few things tomorrow. I am pretty sure you also get the same problem with Marlin. It is déjà vu all over again. 

What I would try, but only if I were you, because I like testing these things, I would swap the white cable of the BLTouch with the pin 29 cable of the filament sensor and then ALSO swap the pin assignements in the config file.

Why? Because the filament sensor has "misused" the pin 29, = beeper pin. This has no such BAD capacitor like the z_stop pin. If you put the BLTouch white cable there, it would be better.

And the filament sensor cable (white), you can put in the place of the Z-stop, where the white BLTouch cable was.

witte1000, 2020-04-03:
Ok thanks will try that also,
I use marlin bugfixes 2.0

[Configuration (copy).h.log](https://github.com/KevinOConnor/klipper/files/4428468/Configuration.copy.h.log)
[Configuration_adv (copy).h.log](https://github.com/KevinOConnor/klipper/files/4428469/Configuration_adv.copy.h.log)

KevinOConnor, 2020-04-03:
> can we give him a one-or-two liner to activate SW mode after the deploy?

As far as I know, it's not difficult to test probing in touch mode - it should be something like:

```diff
--- a/klippy/extras/bltouch.py
+++ b/klippy/extras/bltouch.py
@@ -136,6 +136,7 @@ class BLTouchEndstopWrapper:
         self.sync_print_time()
         duration = max(MIN_CMD_TIME, self.pin_move_time - MIN_CMD_TIME)
         self.send_cmd('pin_down', duration=duration)
+        self.send_cmd('touch_mode')
         self.send_cmd(None)
     def test_sensor(self):
         if not self.pin_up_touch_triggered:
```

As a reminder, any time changing the source code, be sure to run `sudo service klipper restart`.

-Kevin

FanDjango, 2020-04-03:
@witte1000 I can't re-read the entire thread to be sure, so please don't be frustrated if I ask to be sure: And the probe works ok with Marlin? Yes/No?

FanDjango, 2020-04-03:
@witte1000 @KevinOConnor 
Oh thanks for including the config of Marlin.
I can see:
```
  // Feature: Switch into SW mode after a deploy. It makes the output pulse longer. Can be useful
  //          in special cases, like noisy or filtered input configurations.
  #define BLTOUCH_FORCE_SW_MODE
```

I recommend the touch-mode code change Kevin posted right above before you check any changes to your hardware. Definitely.

witte1000, 2020-04-03:
@FanDjango 

Oh no I'm not frustrated at all :) 
I'm happy to help you guy's to help me and the community ... 
Appreciate it very much !!! 

To answer your question, yes with these setting in marlin everything works like expected, is this hack good is another question but it work, if you try marlin 2.0 without the bugfixes then I cannot upload to the printer because it's to big, that's why i want to try klipper and offcourse also the nice features of klipper :)

Tomorrow I gonna check the touch-mode change out,
Thank you all guy's for the wonderful help !!!
Kind Regards,

FanDjango, 2020-04-04:
@witte1000 On a separate issue, there is a similar case.

I see in your Marlin configs that you need

```
// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS
```

**before you test anything else**, activate the pull-up for your bltouch sensor pin in the klippy config file by putting "^" in front and test with that.

Then in a second step you may check out the additional "touch-mode" command mentioned above.

witte1000, 2020-04-04:
@FanDjango 

I don't think i understand you correctly but it seems I don't have pull-up ??
Which file and line ?
however do have sensor_pin: ^PC4 is this what you mean ?

Adding self.send_cmd('touch_mode') in the bltouch.py it seems to work, the bed stop raising up, doing this 2 time 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4431274/klippy.log)

FanDjango, 2020-04-04:
Ok, yes, I see that you already have got ^PC4. And touch mode, it now works, good, @KevinOConnor and I will discuss, I suppose. It's an easy quicky change. And you should do some more testing now with this special code.

witte1000, 2020-04-04:
ok, I'm trying now to navigate with the Octoprint control butttons,
The X/Y is working like expected but the Z is not working cannot move the bed up or down, also I hear that the motor is getting power but doesn't move.
However it do move when running g28 correctly ...

[klippy.log](https://github.com/KevinOConnor/klipper/files/4431382/klippy.log)