Klipper3d/klipper #2323: [feature request] Mirror / Duplication Mode for IDEX - Dual-X

BAUBLITZ, 2019-12-27:
I miss an Mirror and Duplication Mode in Klipper for my Tenlog TL-D3S.
The mirror and duplication mode halves the size of the print bed in both modes on x-axis.
in mirror mode, the second printhead performs the movements of the first printhead in reverse, in duplex mode the second printhead performs the same movements as the first printhead.
This naturally halves the printing time of 2 identical objects.
I have the problem in original firmware with print over usb. With marlin 2.0 the modes work fine but the board base on an 8 bit  arduino 2560 with 7 steppers and is to slow for the calculations of movements. even when printing with only one printhead.
With klipper, the printer runs much smoother and faster, and the print quality with klipper is much better, so I would like to stay with klipper.

KevinOConnor, 2019-12-30:
This looks like a dup of #1131. 

-Kevin

BAUBLITZ, 2020-01-11:
Yes , but you have set the #1131 to inactive and i have not found any information for the feature in klipper. Is there any chance to implement this feature in klipper ? The printquality with Klipper is better than all other Firmwares i have tested.

closed 2020-01-11 by BAUBLITZ

KevinOConnor, 2020-01-11:
Thanks.  Alas, it would require a developer willing to implement and submit it.  It's not something I, personally, will have time to work on in the near future.

-Kevin

nailgilaziev, 2020-05-19:
sad, Idex is a very interesting thing. 
And klipper is one of the alternative come to mind, when you search alternatives to RepRapFirmware on duet.

kondziogithub, 2020-12-18:
> ... The printquality with Klipper is better than all other Firmwares i have tested ...

@BAUBLITZ can you share your printer config for Tenlog TL-D3 Pro?  I am in the process of configuring mine and cannot get things going. Thanks.

BAUBLITZ, 2020-12-23:
You can download my config from here https://github.com/tenlog/TL-D3/files/5033759/printer.zip it for an Tenlog-D3S.

Look here https://github.com/tenlog/TL-D3/issues/7