Klipper3d/klipper #2324: [feature request] PID gain scheduling

bjones14, 2019-12-27:
It would be helpful to have the ability to change the PID gain values on a schedule.  Perhaps a command similar to loading and saving bed meshes that can be used for PIDs?  I'm not sure if something like this exists...

My use case would be using a different PID to get to temperature vs maintaining temperature, and perhaps a different PID for certain conditions such as cooling fan on/off.

The ability to schedule/modify PID gains on-the-fly would add a lot of flexibility and greater control over temperatures.

KevinOConnor, 2019-12-30:
This looks like a dup of #1167 .

-Kevin

KevinOConnor, 2020-01-11:
I'm going to close this in favor of the existing #1167 

-Kevin

closed 2020-01-11 by KevinOConnor