Klipper3d/klipper #2325: Bltouch z_offset configuration

IgorKokovkin, 2019-12-27:
I quite new in klipper. Tried to resolve that thing:
When i issue g28 and then g1 z=0. I have distance bw table and nozzle more then 0.3mm.
Please give me advice. Thank you.

Tried to calibrate using PROBE_CALIBRATE, got strange result:

Send: TESTZ Z=-
Recv: // Z position: ?????? --> 2.800 <-- 3.000
Recv: ok
[...]

[...]
Send: TESTZ Z=+.1
Recv: // Z position: 0.600 --> 0.700 <-- 0.800
Recv: ok
[...]
Send: ACCEPT
Recv: // bltouch: z_offset: -0.700
Recv: // The SAVE_CONFIG command will update the printer config file
Recv: // with the above and restart the printer.

saved config, restarted klipper. Got that error:

Send: G28
Recv: // position_endstop in section 'stepper_z' must be between position_min and position_max
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! position_endstop in section 'stepper_z' must be between position_min and position_max

BlackStump, 2019-12-28:
are you following the steps to the letter ?
https://www.klipper3d.org/Probe_Calibrate.html
and here for the paper test
https://www.klipper3d.org/Bed_Level.html#the-paper-test

this command is missing a amount
>Send: TESTZ Z=-

Bltouch has a positive offset.  -0.700 is telling klipper the bltouch triggers 0.700mm below your bed surface - usual bltough offset is around 1.0~1.6mm depending on version.
>Recv: // bltouch: z_offset: -0.700

https://www.antclabs.com/manual

ld3300, 2019-12-28:
Will you upload your klippy log file?

IgorKokovkin, 2019-12-28:
I have attached it to the previous message, will do again.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4006770/klippy.log)

BlackStump, 2019-12-28:
can you enlighten me why this is needed?
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L457
and why you would need this in your homing overide?
>SET_KINEMATIC_POSITION 
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#force-movement

I am curious why diagnostic commands are used in a homing_overide. If it was me I would keep things as simple as they need to be to get the job done especially when you have a problem or things are not working as expected.
maybe try the example here ie simple you can always pimp it later.
https://github.com/KevinOConnor/klipper/blob/master/config/sample-probe-as-z-endstop.cfg#L23

IgorKokovkin, 2019-12-28:
I used example from printer community, but unfortunately no one from that community was unable explain the solution.
understood your idea. Will check it shortly. 
agree, SET_KINEMATIC_POSITION is not used for its intended purpose in this case

IgorKokovkin, 2020-01-22:
Completed changes, works fine. Thank you.

closed 2020-01-22 by IgorKokovkin