Klipper3d/klipper #2326: Max31856: Cold Junction High Fault

JeffDM2, 2019-12-28:
I'm having a problem getting a thermocouple amplifier working.

There are five MAX31856 on this board. So far I've tested all five and they all have the same results when the thermocouple is missing (Max31856: Thermocouple Range Fault) and when there's a thermocouple connected (Max31856: Cold Junction High Fault).

I compared the schematics against those of Duet's thermocouple expansion board (they use the same chip as on this board) and those in the data sheet and all that seems correct. I don't have an obvious way to test if the surrounding components for this amplifier are correct.

I've checked that the SPI0 bus works, same board has TMCs on the same SPI bus and did a DUMP_TMC and it returned the expected register dump. But this is only when I turned off the TC amplifier configuration as I couldn't operate the board when the TC amp was turned on.

Is there something else I can check? I don't know how to read the SPI messages that are in the log file. Everything else on this board seems to work fine so far.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4006477/klippy.log)

KevinOConnor, 2019-12-30:
I'm not sure why you are getting those errors.  As far as I know, the max31856 code does work.  What micro-controller pins did you actually connect to the max31856 sensor?

-Kevin

JeffDM2, 2019-12-30:
I am using spi0 on SAM3X8E. Because there's five of the amps on this board, each one of these is a CS pin, one per amp: PC27, PB9, PA6, PA2, PB17

Relevant config section:

[heater_generic chamber]
gcode_id:		C
heater_pin:		PC24
sensor_type:		MAX31856
spi_bus:		spi0
spi_speed:		2000000
sensor_pin:		PB17
tc_type:		K
tc_use_50Hz_filter:	False
tc_averaging_count:	1
control:		watermark
min_temp:		0
max_temp:		80

JeffDM2, 2019-12-30:
spi_speed was me testing on a lark, I left that at default for most of the tests.

JeffDM2, 2019-12-31:
I've worked the problem a little bit with someone on the Klipper Discord.

It might be a SPI bus conflict. The board uses a buffer on the SPI bus and the schematic notes there can be a bus conflict in trying to use the SPI bus with external devices, _maybe_ it's related. It's the only lead I have right now. As far as I know, this SPI bus has 8 TMCs, 5 thermocouple amps and a flash ROM. And it's fanned out using buffer chips. I might have to just wait until the board manufacturer can get back to me for comment.

Board schematic:
https://www.dropbox.com/s/uhzkf6w57hoiimb/archimajor_0.1b_schematic.pdf?dl=0

Possibly relevant notes circled on schematic:
https://media.discordapp.net/attachments/601514402746859531/661639126654648321/unknown.png

KevinOConnor, 2019-12-31:
> this SPI bus has 8 TMCs, 5 thermocouple amps and a flash ROM.

FYI, if you have multiple chips on an SPI bus then you must correctly configure every chip in the Klipper config file.  If a chip is on the bus and not configured, it may respond to messages not intended for it and thus corrupt the response from the intended chip.  I suggest you physically remove any chip not in active use.  If that's not an option, an alternative is to identify all the CS pins of extra chips and manually configure them in a "high" state (for example, via a static_digital_output config section).

-Kevin

JeffDM2, 2020-01-01:
OK so:

[static_digital_output onboard_led]
pins: PC1, !PC3, PB9, PA6, PA2, PB17

PB9, PA6, PA2, & PB17 are CS to the amps I'm not setting up at the moment. Is that correct? Does it matter where in the config file it's put? It doesn't seem to make a difference.

Alternatively, how to I set up a sensor and not assign it a heater or fan?

JeffDM2, 2020-01-02:
I've set up everything as best as I know how. I'm still getting the cold junction high error.

All thermocouple amps have K type thermocouples hooked up, but that shouldn't be the problem anyway.

I cleaned up my klippy.log to show the config up until the end of the errors.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4017680/klippy.log)

Here is my config file as-is:

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4017681/printer.cfg.txt)

KevinOConnor, 2020-01-11:
Alas, I don't have ideas on what the source of the error is.  It does sound like a hardware issue.  Best advice I'd have would be to try and wire up just one sensor to a dedicated spi bus to verify everything works.

-Kevin

closed 2020-02-02 by KevinOConnor