Klipper3d/klipper #2327: Feature request: Allow simultaneous movement of manual steppers

Spacemarine2018, 2019-12-28:
Manual steppers can only be moved one at a time. For certain applications it would be desireable to move two or more manual steppers at the same time.

The simplest ways would probably to allow two stepper names to be passed to the MANUAL_STEPPER command, like this:

MANUAL_STEPPER STEPPER=firststepper MOVE=30 STEPPER=secondstepper MOVE=55

The movements won't be synchronized, so the steppers might finish at different times, but often this is not a problem since each movement can be hand-tuned with the SPEED ACCEL options.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4006881/klippy.log)

KevinOConnor, 2019-12-30:
FYI, this was discussed a bit in issue #897.

-Kevin

KevinOConnor, 2020-01-23:
I'm going to mark this as inactive as it doesn't appear anyone is currently implementing it.  Should someone be interested, feel free to implement and submit.

-Kevin

closed 2020-01-23 by KevinOConnor

KevinOConnor, 2020-03-25:
FYI, basic support for this was added as of commit 6be552f.

-Kevin