Klipper3d/klipper #2328: CNC branch out there?

water-rhatt, 2019-12-28:
I have seen and read previous CNC related post and understand, and appreciate, @KevinOConnor not wanting to put valuable time into adding milling support. 

That said, there is a lot to like about Klipper for CNC router machines. Leveraging a more powerful 32bit controller and treating the Arduino as a dumb relay is something many bench top and pro-consumer machines could use. 

Unfortunately I am not a Python programmer, or I would offer to try to create a milling branch. Mostly it appears that Klipper is missing WCS and TRO support. Perhaps also support for certain canned motions like helical and drilling/tapping. 

Is there a milling branch already out there? I would be happy to help test on my machine and even a little coding if someone out there has taken the lead on this. 

Thanks.

Spacemarine2018, 2019-12-29:
I'm using Klipper also for something completely different, It's only CNC in the widest sense, since I'm bulding an automated pen plotter and nailgun.

I ran into quite a few limitations that were implemented to safeguard against errors made by 3D-printer users, which don't make sense in a CNC machining context. Maybe we can start a list of things that a special CNC branch should take care of? Maybe some of them are easy to implement.

- Every axis should have an individual speed and acceleration limit. At the moment X and Y operate under the same limits and additionally Z is limited to a value lower than X or Y. [https://github.com/KevinOConnor/klipper/issues/2166](url) and [https://github.com/KevinOConnor/klipper/issues/2155](url)

- Parameters like square-corner velocity should removed (or set to zero as standard) as they change the speed and acceleration in favor of a cleaner extrusion, which can have an impact on tool-path.

yhaiovyi, 2020-01-07:
@Spacemarine2018 your first bullet point makes perfect sense in 3d printing world too. And should be relatively easy to implement. As for a second one, I don't see any problems to set is to 0, if that's what you want.

Spacemarine2018, 2020-01-08:
@yhaiovyi 

I also think it wouldn't be too hard to implement, at least the decoupling of Z from X and Y, which I already requested here: https://github.com/KevinOConnor/klipper/issues/2166

Decoupling of X and Y is probably not so easy, especially for non-cartesian systems like CoreXY, which I also already requested: https://github.com/KevinOConnor/klipper/issues/2155

KevinOConnor, 2020-02-02:
I'm going to close this as it looks like the conversation has concluded.  If anyone has cnc specific code that they wish to see included, then feel free to open a PR with it and I'll review.  CNC isn't necessarily a focus, but I would guess we could include cnc type features in to the main repo.

-Kevin

closed 2020-02-02 by KevinOConnor

water-rhatt, 2020-02-05:
Yes, it is a shame.  There are few 32bit solutions for CNC.  But using Klipper I can see there is still work to bring it up to the functionality of Marlin, and that is the main focus.  Grbl would actually be an easier port since it has fewer non-standard g-code functions.  But that will have to wait for another project.  No doubt as Klipper matures there will be more interest.

Thanks!