Klipper3d/klipper #2330: Ender 5 Plus (Creality V2.2 PCB)

TheRealAlexV, 2019-12-28:
I have an Ender 5 Plus (350mm build surface, BLTouch, Touch Screen Interface) that I would like to use klipper with, however I'm not entirely sure what configuration is needed for the Ender 5 Plus. It's currently using the Creality V2.2 PCB. 

In about 2 weeks, I'll be installing an upgraded Creality V2.2 Silent Board with the TMC2208 motor drivers. Would I need to change anything once that board is in? As I understand, it is the same exact Creality V2.2 PCB just with the TMC2208 drivers.

I noticed an example config for the Ender 5 at https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender5-2019.cfg, however I don't know if this will be compatible, since as far as I'm aware, the Ender 5(Non Plus) is not a 2.2 board.

jakep82, 2019-12-28:
I don't believe Creality connects the UART from the 2208 drivers to available pins on the MCU. That means the board won't be compatible with Klipper. See the Klipper FAQ for information about why 2208 drivers must be configured with UART enabled.

Hywelmartin, 2019-12-29:
probably the same config as the rest of ender machines.. ender 3, 5.. different size and different placement of endstops..

KevinOConnor, 2020-01-23:
I'm going to close this as it looks like the question was answered as best as we could.  (The Creality boards with tmc2208 and no uart wiring are not supported by Klipper; the other boards should be possible to configure by using one of the other boards as reference.)

-Kevin

closed 2020-01-23 by KevinOConnor