Klipper3d/klipper #2331: Feature Req: Set PID values per G-Codes or have different values for different set points

ProfDrYoMan, 2019-12-29:
Hi,

I print with 190 C lowest and up to 260 C highest. PID values are different (significant) for these two set-points. Right now I use one PID set which is measured at 210 C.

Yes, it is „ok“ but it would be better to be able to either set the PID values per G-Code (Macro) or have defined areas for PID values.

Let’s say something like that.

```
[PID-170-190]
P=xxx
I=xxx
D=xxx
[PID-190-220]
...
[PID-220-270]
...
```

Spacemarine2018, 2019-12-29:
This would definately make sense. It seems this request is very related to this one:
https://github.com/KevinOConnor/klipper/issues/2324

bjones14, 2019-12-30:
> 
> 
> This would definately make sense. It seems this request is very related to this one:
> #2324

Right, this is just PID gain scheduling, and using temperature or cooling fan as scheduling variables.

KevinOConnor, 2019-12-30:
This looks like a dup of #1167.

-Kevin

KevinOConnor, 2020-01-23:
I'm going to close this in favour of the existing #1167. 

-Kevin

closed 2020-01-23 by KevinOConnor