Klipper3d/klipper #2332: Feature Req: Not only Ziegler-Nichols for PID calibrations

ProfDrYoMan, 2019-12-29:
Hi,

Ziegler-Nichols is used for PID calibration.

Chien, Hrones and Reswick should get better results for disturbance value driven controlled systems (which is the case for the heater)

For aperiodic disturbance driven controlled system these calculations should get better results.

Kp = 0.95 * (Tg / (Tu * Ks)

Tn = 2.4 * Tu

Tv = 0.42 * Tu

See https://de.wikipedia.org/wiki/Faustformelverfahren_(Automatisierungstechnik)

Sorry, did not find an English version of that.

Redeem (Another FW for the Replicape board on BBB) did support that and I did get better results tracking the temperature if the fans start/stop or the extruder flow was changing significant.

tonn0297, 2019-12-29:
i can 2. this. there needs to be a better way to pid. the current pid method works but is not acurate or stable enugh

ProfDrYoMan, 2019-12-29:
The default smooth_time of 2 sec is also way to much.

Are there really that bad temp sensor readings out there.

Anything above 0.1 s will be useless for proper and fast PID.

ProfDrYoMan, 2019-12-30:
Another (easier) option would be Tyreus-Luyben. As far as I remember that was default in Redeem.

KevinOConnor, 2019-12-30:
FYI, the PID_CALIBRATE command does report the underlying Ku and Tu values in the klipper log file.  (As well as all the underlying temperature readings if the WRITE_FILE option is used.)  So, it is possible to use that tool and then manually calculate a different PID using some other formula.

FWIW, my experience with the PID is that one can spend huge amounts of time calibrating it and yet never achieve "excellent" results.  I suspect "adequate" results is the only achievable goal with the PID.  As such, the extremely common Ziegler-Nichols / Astrom-Hagglund seems sufficient to me.

Separately, in the past I've spent time on a first-order-plus-delay-time (fopdt) based heater control implementation.  You can still find that code on the work-fopdt-20180405 branch.  It doesn't work particularly well, however.

Beyond that, if there is interest in extending this code, a developer would need to implement and submit it.

Cheers,
-Kevin

ProfDrYoMan, 2019-12-31:
I spent some time to investigate. Calculated TL method PID coefficients by hand.

Result: Not as good as ZN coefficients. By far. Funny, since this got me better results in Redeem, than ZN.

Since the process is asymmetric, heating faster than cooling, it is hard to find proper PID values. A simple PID does not match.

Maybe on simple option could enhance the behavior when enabling/disabling fans.

How about adding an option which directly adds to the fan output PWM. If fan is running 100% it will be added up 100%. If the fan is running less, the additional value should be accordingly.

Because stationary my temperature is rock solid within 0.5C only if the fan kicks in it deviates by 2C or more.

BlackStump, 2019-12-31:
@ProfDrYoMan  do you have insulation on your hotend? if not it is a good option.
Fan outputs should not be directed at the hotend instead it should be focused at the filament leaving the nozzle.

ProfDrYoMan, 2020-01-01:
The hotend is insulated with a silicon sock and the fan is guided blowing onto the tip of the nozzle (as good as possible).
Cannot put this guidance any lower.

KevinOConnor, 2020-01-03:
FYI, I generally think of fan based heater changes as an example of a "feed forward PID".  That was raised in #597.  Also, #1167 might be another approach.

-Kevin

bjones14, 2020-01-04:
I would like to work on developing this PID Gain Scheduling/Feed Forward and submit a request when it is completed.  I think there are a lot of good ideas from all these different issues which should be a good base for an implementation that supports all of these features.

Coffee0297, 2020-01-08:
i also see difference in the response time of the pid loop... idle its fine but when "printing" it goes bad, maybe add a parameter so we can decide how long each cycle of pid should take. 
-Tonny

Coffee0297, 2020-01-12:
i can see marlin uses Ziegler-Nichols 'Classic' method. so there must be somthing that makes klipper not be as good/precise as marlin. 
@KevinOConnor heres a link to how marlin do it, and please see previus comment.
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/temperature.cpp#L341
-Tonny

KevinOConnor, 2020-02-14:
Okay, I'm going to mark this as inactive.  If a developer is interested in doing development and testing I'd certainly be interested in hearing about the results.

-Kevin

closed 2020-02-14 by KevinOConnor

bjones14, 2020-02-15:
I just submitted a pseudo-related pull request to this issue - https://github.com/KevinOConnor/klipper/pull/2507

Considering that, I would also like to work on this request.  It sounds like a parameter in the printer.cfg could be added easily enough that would allow the end-user to choose which algorithm/method they want to use to calculate their terms?

If that's the case, anyone interested in creating a snippet of Python code that does that, it would be real helpful that way I can just take and place it in the code where it needs to be.