Klipper3d/klipper #2334: Internal error on command:"G1" (when print with T1, the second extruder)

robustini, 2019-12-29:
When i'm printing with the second extruder sometimes it stop, while with the first extruder the same gcode he never had a problem during a lot of test.
The error is "Internal error on command:"G1", see the log:

[klippy.log](https://github.com/KevinOConnor/klipper/files/4008233/klippy.log)

KevinOConnor, 2019-12-30:
It doesn't look like the attached log file is complete.  We'll need the full log in order to assist.  Also, it looks like you've disable the max_extrude_cross_section check - please reproduce the issue without that check disabled.

-Kevin

robustini, 2019-12-31:
Here the full log...

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4012216/klippy.log.zip)

robustini, 2019-12-31:
I removed the "max_extrude_cross_section" in the extruder1 section and now it seems that printing continues without problems.

robustini, 2019-12-31:
Yes, it seems that the problem is related to "max_extrude_cross_section", if disabled it produces this problem.

closed 2019-12-31 by robustini